SpenceKonde/ATTinyCore #214: stk500 support?

tedder, 2018-05-04:
Can you add a tinycore programmer profile for stk500? It appears stk500 is different than stk500v1 or stk500v2. I've found I can attempt to upload code with one of the latter entries, which will fail, then I can copy/paste the avrdude command and change it to `-cstk500` and it works.

SpenceKonde, 2018-05-04:
This is mysterious, can you provide a bit more information about this?
What part are you programming?
What programmer are you using?
What is the full text of the error message you get? (ideally, verbose output)

Thanks

tedder, 2018-05-04:
Sure Spence. I'm using Azzy's breakout for the ATTINY84 (with an 84A) in it. Programmer is [the 2.1 from pololu](https://www.pololu.com/product/3172). I'm on a modern Mac (touchbar macbook pro, macos 10.13.4).

Update while dumping logs- looks like the issue is actually that stk500v2 comes with the `-Pusb` param. If I replace it with the serial port it works. So `stk500v2` and `stk500` work as long as I specify the port, `stk500v1` doesn't work. Which means the problem is that the two supported options don't have that combination.

```
$ /Applications/Arduino_183.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/ted/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.5/avrdude.conf -v -v -D -pattiny84 -cstk500v1 -P/dev/cu.usbmodem00221792 -Uflash:w:/nofile:i

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/ted/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.5/avrdude.conf"
         User configuration file is "/Users/ted/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem00221792
         Using Programmer              : stk500v1
         Setting bit clk period        : 5.0
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
^C
```

```
$ /Applications/Arduino_183.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/ted/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.5/avrdude.conf -v -pattiny84 -cstk500v2 -Pusb -Uflash:w:/var/folders/8q/zb610w691_j2rl78z_npp93m0000gn/T/arduino_build_713660/Blink.ino.hex:i

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/ted/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.5/avrdude.conf"
         User configuration file is "/Users/ted/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : stk500v2
         Setting bit clk period        : 5.0
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.
```

Now keeping it as v2 but setting the port:

```
$ /Applications/Arduino_183.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/ted/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.5/avrdude.conf -v -pattiny84 -cstk500v2 -P/dev/cu.usbmodem00221792 -Uflash:w:/var/folders/8q/zb610w691_j2rl78z_npp93m0000gn/T/arduino_build_713660/Blink.ino.hex:i

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/ted/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.5/avrdude.conf"
         User configuration file is "/Users/ted/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem00221792
         Using Programmer              : stk500v2
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny84
         Chip Erase delay              : 15000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
[... 27 lines cut ...]
         Vtarget         : 4.7 V
         SCK period      : 8.7 us
         Varef           : 0.0 V
         Oscillator      : Off

avrdude: AVR device initialized and ready to accept instructions

(snipping successful output)
```

I don't think it matters, but here's the product/vendor info for the Pololu programmer:
```
                Pololu USB AVR Programmer v2.1:

                  Product ID: 0x00bb
                  Vendor ID: 0x1ffb
                  Version: 1.02
                  Serial Number: 00221798
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Pololu Corporation
                  Location ID: 0x00111000 / 24
                  Current Available (mA): 500
                  Current Required (mA): 100
                  Extra Operating Current (mA): 0
```

SpenceKonde, 2018-05-04:
Goddamnit, yet another bloody programmer, incompatible with all the programmer definitions I supplied. I'll add the STK500 development board programmer definition...

closed 2018-05-04 by SpenceKonde

tedder, 2018-05-04:
hah. The joy of a combinatorial explosion.

SpenceKonde, 2018-05-04:
The version in github should work now; It will not be in the board manager version until 1.1.6 is ready (it's got a ways to go - 1.1.5 was released recently). 

In the meantime, the one that comes with the AVR board package (ie, the one not marked (ATTinyCore)) will work with the tiny44/84/45/85 - the lousy avrdude.conf that the IDE comes with at least has definitions for those parts. It lacks definitions for all the more exotic attiny parts though.

tedder, 2018-05-04:
Aha, thanks. Nice to know 'tinycore' isn't needed for the x4 series.

Understand about the naming choices too (on the git commit).

commit 3213cce7 (2018-05-04): Fix #214

Support yet another ISP programmer.