SpenceKonde/ATTinyCore #216: Error build SPI library for ATTiny1634

colesnicov, 2018-05-25:
Hi guys. I've got the problem:

```

Compiling library "SPI"
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp" -o "/tmp/arduino_build_431784/libraries/SPI/SPI.cpp.o"
In file included from /home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/avr/include/avr/io.h:99:0,
                 from /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.h:27,
                 from /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:19:
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp: In static member function 'static void tinySPI::begin()':
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern/pins_arduino.h:126:21: error: 'DDC1' was not declared in this scope
 #define USCK_DD_PIN DDC1
                     ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:263:25: note: in expansion of macro 'USCK_DD_PIN'
     USI_SCK_PORT |= _BV(USCK_DD_PIN);   //set the USCK pin as output
                         ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern/pins_arduino.h:127:19: error: 'DDB2' was not declared in this scope
 #define DO_DD_PIN DDB2
                   ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:264:25: note: in expansion of macro 'DO_DD_PIN'
     USI_DDR_PORT |= _BV(DO_DD_PIN);     //set the DO pin as output
                         ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern/pins_arduino.h:128:19: error: 'DDB1' was not declared in this scope
 #define DI_DD_PIN DDB1
                   ^
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:265:26: note: in expansion of macro 'DI_DD_PIN'
     USI_DDR_PORT &= ~_BV(DI_DD_PIN);    //set the DI pin as input
                          ^
Použití knihovny SPI ve verzi 1.0 v adresáři: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI 
exit status 1
Nastala chyba při kompilaci u desky ATtiny1634 (Optiboot).
```
Try to compile standard _[SPI example from coreboard](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino)_ for **attiny1634**
**ATTinyCore version 1.1.5**

SpenceKonde, 2018-05-25:
Will investigate.

colesnicov, 2018-05-27:
Hello. I will create video and share this for you.  It will be easier to understand. I can not speak English very well. [this link](https://youtu.be/drNbXtQIRE4)

SpenceKonde, 2018-05-28:
When I compile that same example with the same settings, the verbose compile output is

```

C:\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\arduino-1.8.5\hardware -hardware C:\Users\Pete\Documents\Arduino\hardware -tools C:\arduino-1.8.5\tools-builder -tools C:\arduino-1.8.5\hardware\tools\avr -built-in-libraries C:\arduino-1.8.5\libraries -libraries C:\Users\Pete\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attiny1634opti:LTO=disable,clock=8internal,eesave=aenable,bod=1v8,bodact=disabled,bodpd=disabled -ide-version=10805 -build-path C:\Users\Pete\AppData\Local\Temp\arduino_build_849606 -warnings=none -build-cache C:\Users\Pete\AppData\Local\Temp\arduino_cache_578648 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI\examples\BarometricPressureSensor\BarometricPressureSensor.ino
C:\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\arduino-1.8.5\hardware -hardware C:\Users\Pete\Documents\Arduino\hardware -tools C:\arduino-1.8.5\tools-builder -tools C:\arduino-1.8.5\hardware\tools\avr -built-in-libraries C:\arduino-1.8.5\libraries -libraries C:\Users\Pete\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attiny1634opti:LTO=disable,clock=8internal,eesave=aenable,bod=1v8,bodact=disabled,bodpd=disabled -ide-version=10805 -build-path C:\Users\Pete\AppData\Local\Temp\arduino_build_849606 -warnings=none -build-cache C:\Users\Pete\AppData\Local\Temp\arduino_cache_578648 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI\examples\BarometricPressureSensor\BarometricPressureSensor.ino
Using board 'attiny1634opti' from platform in folder: C:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr
Using core 'tinymodern' from platform in folder: C:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr
Detecting libraries used...
"C:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\cores\tinymodern" "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\sketch\BarometricPressureSensor.ino.cpp" -o "nul"
"C:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\cores\tinymodern" "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI" "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\sketch\BarometricPressureSensor.ino.cpp" -o "nul"
"C:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\cores\tinymodern" "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI" "C:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI\SPI.cpp" -o "nul"
Generating function prototypes...
"C:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\cores\tinymodern" "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI" "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\sketch\BarometricPressureSensor.ino.cpp" -o "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\arduino-1.8.5\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\cores\tinymodern" "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI" "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\sketch\BarometricPressureSensor.ino.cpp" -o "C:\Users\Pete\AppData\Local\Temp\arduino_build_849606\sketch\BarometricPressureSensor.ino.cpp.o"
Compiling libraries...
Compiling library "SPI"
"C:\arduino-1.8.5\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\cores\tinymodern" "-IC:\Users\Pete\Documents\Arduino\hardware\ATTinyCore\avr\libraries\SPI" "C:\Users\P [... cut ...]
Global variables use 279 bytes (27%) of dynamic memory, leaving 745 bytes for local variables. Maximum is 1024 bytes.

```

What version of Arduino Board Package (see tools -> boards manager) do you have installed? 

I suspect a problem with the Arduino IDE installation/environment, not specific to attinycore. I would uninstall IDE, blow away Arduino15 folder, reinstall. and see if that resolved problems. I did not like the look of all those strange warnings going by ( points to a larger problem with your IDE installation)

colesnicov, 2018-05-28:
If you think "Arduino AVR Boards", it is in version 1.6.21. Todle is the whole output of the compilation:

```
/home/denis/opt/arduino/arduino-builder -dump-prefs -logger=machine -hardware /home/denis/opt/arduino/hardware -hardware /home/denis/.arduino15/packages -hardware /home/denis/Arduino/hardware -tools /home/denis/opt/arduino/tools-builder -tools /home/denis/opt/arduino/hardware/tools/avr -tools /home/denis/.arduino15/packages -built-in-libraries /home/denis/opt/arduino/libraries -libraries /home/denis/Arduino/libraries -fqbn=ATTinyCore:avr:attiny1634opti:LTO=disable,clock=8internal,eesave=aenable,bod=1v8,bodact=disabled,bodpd=disabled -ide-version=10805 -build-path /tmp/arduino_build_331773 -warnings=more -build-cache /tmp/arduino_cache_352504 -prefs=build.warn_data_percentage=75 -verbose /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino
/home/denis/opt/arduino/arduino-builder -compile -logger=machine -hardware /home/denis/opt/arduino/hardware -hardware /home/denis/.arduino15/packages -hardware /home/denis/Arduino/hardware -tools /home/denis/opt/arduino/tools-builder -tools /home/denis/opt/arduino/hardware/tools/avr -tools /home/denis/.arduino15/packages -built-in-libraries /home/denis/opt/arduino/libraries -libraries /home/denis/Arduino/libraries -fqbn=ATTinyCore:avr:attiny1634opti:LTO=disable,clock=8internal,eesave=aenable,bod=1v8,bodact=disabled,bodpd=disabled -ide-version=10805 -build-path /tmp/arduino_build_331773 -warnings=more -build-cache /tmp/arduino_cache_352504 -prefs=build.warn_data_percentage=75 -verbose /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino
Using board 'attiny1634opti' from platform in folder: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5
Using core 'tinymodern' from platform in folder: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5
Detecting libraries used...
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "/tmp/arduino_build_331773/sketch/BarometricPressureSensor.ino.cpp" -o "/dev/null"
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/tmp/arduino_build_331773/sketch/BarometricPressureSensor.ino.cpp" -o "/dev/null"
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp" -o "/dev/null"
Generating function prototypes...
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/tmp/arduino_build_331773/sketch/BarometricPressureSensor.ino.cpp" -o "/tmp/arduino_build_331773/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/denis/opt/arduino/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_331773/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/denis/.arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny1634 -DF_CPU=8000000L -DARDUINO=10805 -DARDUINO_AVR_ATTINY1634_OPTIBOOT -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tinymodern" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/tmp/arduino_build_331773/sketch/BarometricPressureSensor.ino.cpp" -o "/tmp/arduino_build_331773/sketch/BarometricPressureSensor.ino.cpp.o"
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino: In function 'void loop()':
[... cut ...]
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp:265:26: note: in expansion of macro 'DI_DD_PIN'
     USI_DDR_PORT &= ~_BV(DI_DD_PIN);    //set the DI pin as input
                          ^
Using library SPI at version 1.0 in folder: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI 
exit status 1
Error compiling for board ATtiny1634 (Optiboot).
```

As for installation Arduino IDE, so I'm sure I've got the whole Linux fucked ...

So I will try to reinstall the IDE.

colesnicov, 2018-05-29:
You're right. After updating the IDE, everything is in order.

Output:
```
Archiving built core (caching) in: /tmp/arduino_cache_937794/core/core_ATTinyCore_avr_attiny1634opti_LTO_disable,clock_8internal,eesave_aenable,bod_1v8,bodact_disabled,bodpd_disabled_61a90b56cfd455750ef25700d7d301a2.a
Projekt zabírá 6962 bytů (44%)  úložného místa pro program. Maximum je 15744 bytů.
Globální proměnné zabírají 278 bytů (27%)  dynamické paměti, 746 bytů zůstává pro lokální proměnné. Maximum je 1024 bytů.
```

closed 2018-05-29 by colesnicov