SpenceKonde/ATTinyCore #218: Error compiling ArduinoSPI

colesnicov, 2018-06-04:
Hello. I'm trying to compile a standard example ArduinoSPI because I can program the ATMEGA328 with the help attiny85 but I get the following error:

```
/home/denis/opt/arduino/arduino-builder -dump-prefs -logger=machine -hardware /home/denis/opt/arduino/hardware -hardware /home/denis/.arduino15/packages -hardware /home/denis/Arduino/hardware -tools /home/denis/opt/arduino/tools-builder -tools /home/denis/opt/arduino/hardware/tools/avr -tools /home/denis/.arduino15/packages -built-in-libraries /home/denis/opt/arduino/libraries -libraries /home/denis/Arduino/libraries -fqbn=ATTinyCore:avr:attinyx5:LTO=disable,TimerClockSource=default,chip=85,clock=8internal,eesave=aenable,bod=disable -ide-version=10804 -build-path /tmp/arduino_build_608236 -warnings=default -build-cache /tmp/arduino_cache_907317 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_27005/ArduinoISP.ino
/home/denis/opt/arduino/arduino-builder -compile -logger=machine -hardware /home/denis/opt/arduino/hardware -hardware /home/denis/.arduino15/packages -hardware /home/denis/Arduino/hardware -tools /home/denis/opt/arduino/tools-builder -tools /home/denis/opt/arduino/hardware/tools/avr -tools /home/denis/.arduino15/packages -built-in-libraries /home/denis/opt/arduino/libraries -libraries /home/denis/Arduino/libraries -fqbn=ATTinyCore:avr:attinyx5:LTO=disable,TimerClockSource=default,chip=85,clock=8internal,eesave=aenable,bod=disable -ide-version=10804 -build-path /tmp/arduino_build_608236 -warnings=default -build-cache /tmp/arduino_cache_907317 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_27005/ArduinoISP.ino
Using board 'attinyx5' from platform in folder: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5
Using core 'tiny' from platform in folder: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5
Detecting libraries used...
"/home/denis/opt/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10804 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tiny" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/variants/tinyX5" "/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp" -o "/dev/null"
"/home/denis/opt/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10804 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tiny" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/variants/tinyX5" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp" -o "/dev/null"
Using cached library dependencies for file: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI/SPI.cpp
Generating function prototypes...
"/home/denis/opt/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics   -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10804 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tiny" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/variants/tinyX5" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp" -o "/tmp/arduino_build_608236/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/denis/opt/arduino/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_608236/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/denis/opt/arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD  -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10804 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR   "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tiny" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/variants/tinyX5" "-I/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI" "/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp" -o "/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp.o"
Compiling libraries...
Compiling library "SPI"
Using previously compiled file: /tmp/arduino_build_608236/libraries/SPI/SPI.cpp.o
Compiling core...
Using precompiled core
Linking everything together...
"/home/denis/opt/arduino/hardware/tools/avr/bin/avr-gcc"  -Os  -Wl,--gc-sections -mmcu=attiny85  -o "/tmp/arduino_build_608236/ArduinoISP.ino.elf" "/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp.o" "/tmp/arduino_build_608236/libraries/SPI/SPI.cpp.o" "/tmp/arduino_build_608236/../arduino_cache_907317/core/core_ATTinyCore_avr_attinyx5_LTO_disable,TimerClockSource_default,chip_85,clock_8internal,eesave_aenable,bod_disable_61a90b56cfd455750ef25700d7d301a2.a" "-L/tmp/arduino_build_608236" -lm
/tmp/arduino_build_608236/../arduino_cache_907317/core/core_ATTinyCore_avr_attinyx5_LTO_disable,TimerClockSource_default,chip_85,clock_8internal,eesave_aenable,bod_disable_61a90b56cfd455750ef25700d7d301a2.a(TinySoftwareSerial.cpp.o): In function `getch':
/home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/cores/tiny/TinySoftwareSerial.cpp:70: multiple definition of `getch'
/tmp/arduino_build_608236/sketch/ArduinoISP.ino.cpp.o:/tmp/arduino_modified_sketch_27005/ArduinoISP.ino:300: first defined here
collect2: error: ld returned 1 exit status
Using library SPI at version 1.0 in folder: /home/denis/.arduino15/packages/ATTinyCore/hardware/avr/1.1.5/libraries/SPI 
exit status 1
Error compiling for board ATtiny25/45/85.
```

SpenceKonde, 2018-06-04:
This issue appears to be because ArduinoISP sketch uses a function with the same name as the internal software serial Serial supplied with this core, getch() - this can be fixed easily (find/replace getch in the sketch with some other name), and I'll consider renaming the function in the internal software serial implementation. 

What on earth are you trying to do, though?! It looks like you're compiling ArduinoISP to run on the tiny85 - but the tiny85 doesn't have enough pins for that (you need the three SPI pins, two serial pins, and the pin connected to reset of the target) unless you blew the RSTDSBL fuse. Also, IIRC, the ArduinoISP sketch comes out larger than 8k (not at a system with the IDE on it to check atm), and I'm not sure it works with software serial implementations either!

Usually people put ArduinoISP onto their '328p based arduinos, and use it to upload to their attiny's - not the other way around, which is what you seem to be trying to do.

colesnicov, 2018-06-04:
If I use SoftwareSerial on pines (PB3, PB4), SPI (PB0, PB1, PB2) and RESET (PB5).

As for the size of the compiled sketch, so without SoftwareSerial it is about 4.5kb.
```
         _______
 PB5-|      A     |-VCC
 PB3-|      T     |-PB2
 PB4-|      8     |-PB1
 GND-|      5     |-PB0
      ------------
```

So i try to find / replace :)

colesnicov, 2018-06-04:
I'll let you know if I did it

closed 2018-06-04 by colesnicov