SpenceKonde/ATTinyCore #221: Wire library not functional on ATtiny 88

macdonaldtomw, 2018-06-13:
Compiling the _master_reader_ example for ATtiny88 results in following compilation errors:

`Arduino: 1.8.4 (Windows 10), Board: "ATtiny48/88, Disabled, ATtiny88, 8 MHz (internal), EEPROM retained, B.O.D. Disabled"

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c: In function 'twi_init':

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:74:18: error: 'true' undeclared (first use in this function)

   twi_sendStop = true;  // default value

                  ^

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:74:18: note: each undeclared identifier is reported only once for each function it appears in

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:75:20: error: 'false' undeclared (first use in this function)

   twi_inRepStart = false;

                    ^

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c: In function 'twi_readFrom':

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:164:7: error: 'true' undeclared (first use in this function)

   if (true == twi_inRepStart) {

       ^

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:171:22: error: 'false' undeclared (first use in this function)

     twi_inRepStart = false;   // remember, we're dealing with an ASYNC ISR

                      ^

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c: In function 'twi_writeTo':

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:246:7: error: 'true' undeclared (first use in this function)

   if (true == twi_inRepStart) {

       ^

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:253:22: error: 'false' undeclared (first use in this function)

     twi_inRepStart = false;   // remember, we're dealing with an ASYNC ISR

                      ^

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c: In function '__vector_19':

C:\Users\Boompy\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.5\libraries\Wire\src\twi.c:408:21: error: 'true' undeclared (first use in this function)

    twi_inRepStart = true; // we're gonna send the START

                     ^

exit status 1
Error compiling for board ATtiny48/88.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

SpenceKonde, 2018-06-13:
Hmm, why isn't it picking up the definitions of true and false?

SpenceKonde, 2018-06-21:
Finally had a chance to look at this. This bug is already fixed in the github version and will be fixed in next release ( #212 ). See installation instructions for details on how to install github version. https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

closed 2018-06-21 by SpenceKonde