SpenceKonde/ATTinyCore #223: Preinstantiating Serial objects wastes space

JetIgniter2k, 2018-06-24:
I'm working with the Tiny841 and using optiboot_attiny841at8ser1.hex for my bootloader.  My project doesn't use UASRT0 but a lot of the code for it is compiled and put on the chip anyways.  If I change DEFAULT_TO_TINY_DEBUG_SERIAL to 1 in core_build_options.h, I can save 106 bytes of flash and 31 bytes of RAM.  However, upon inspection of the disassembly, Vector22 is still included.  I would like to free up this additional 106 bytes of flash and 36 bytes of RAM.  If I just comment out lines 107-120 in HardwareSerial.cpp I get this error which is found on line 136:  #error No interrupt handler for usart 0.  

Is there a reason that the interrupt handler for UASRT0 is required?

SpenceKonde, 2018-06-26:
I'm not sure if it's possible to have these only be initialized if the sketch uses Serial0. Does anyone know of a way to achieve this, short of yet another menu in tools to control which hardware serial ports are initialized? I am concerned about the usability implications of this, as I routinely see posts and field inquiries where people report trying every permutation of the menu options when they have a problem (on the "I don't understand this, but maybe it's set wrong). I wish I could create a submenu, so it would be like tools -> Advanced Options -> ... or something, so I could group all the confusing menus into one submenu.

JetIgniter2k, 2018-06-27:
Setting it up as a pound define in core_build_options.h wouldn't be so bad.  Instantiate everything by default and let the advanced users tweak things if they want to.

I suppose you could have a preMain() function in the sketch for the defines that you want to pass to the core libraries and try one of the options [here](https://stackoverflow.com/questions/10897552/call-a-function-before-main). 

I commented out the error on line 136 and it compiles fine but I haven't had a chance to test it out on a chip.  I don't expect there to be any issues, mostly I was just curious why that error was there.

SpenceKonde, 2018-07-15:
If you want to provide a PR that makes it possible to disable a UART with a #define in core_build_options.h, I would welcome that. I have no plans to implement this myself - 

executing code before the rest of main() isnt what we need - it's a define known at compile time, visible to the compiler from the context of the core hardwareserial library. I don't know how to pull that off. 

I would also welcome any other suggestions for reducing the resource use of the hardware serial ports. Maybe I should just reduce the default size of the buffer?

closed 2018-07-15 by SpenceKonde