SpenceKonde/ATTinyCore #224: External Interrupts on ATTiny841

kasparsd, 2018-06-29:
We currently report only pin 1 as a valid interrupt pin:

https://github.com/SpenceKonde/ATTinyCore/blob/e5824581b2adef573e9bffc0af8ebd9b07aa9f57/avr/cores/tinymodern/core_pins.h#L63

However, the datasheet has the following:

> External Interrupts are triggered by the INT0 pin, or by any of the PCINTn pins. Note that, if enabled, the interrupts will trigger even if the INTn or PCINTn pins are configured as outputs. This feature provides a way of generating software interrupts.
> 
> The pin change interrupts trigger as follows:
> - Pin Change Interrupt 0 (PCI0): triggers if any enabled PCINT[7:0] pin toggles
> - Pin Change Interrupt 1 (PCI1): triggers if any enabled PCINT[11:8] pin toggles

Should this be updated to return `0` for pins `PCINT[7:0]` and `1` for pins `PCINT[11:8]`?

SpenceKonde, 2018-06-30:
This is correct behavior. 

In order to maintain predictable behavior, we follow the same behavior as the official core uses for Arduino API functions, including the digitalPinToInterrupt() macro - it only reports full-featured "external interrupt" pins, those numbered INTn where n is a number in Atmel's pin function scheme; these are the pins supported by attachInterrupt() (and this is what this macro is normally used with). Pin Change interrupts are not included. 

Pin change interrupts have more limited functionality, and cannot be used with attachInterrupt(), can only trigger on change (not edge or level), and are triggered in two banks, requiring the ISR to be smart enough to read the pins and figure out what change occurred - though at the same time, with direct port reads, this can be an asset, for example when reading rotary encoders (particularly multiple rotary encoders on the same port).

closed 2018-06-30 by SpenceKonde

kasparsd, 2018-06-30:
Thanks for the detailed explanation @SpenceKonde! Understood!