SpenceKonde/ATTinyCore #225: Universal Wire library error reporting on x41/828

SpenceKonde, 2018-07-15:
With an ATTiny841 (or other part that uses software I2C master), Wire library's requestFrom() will report having received data, even when there is no slave present at the specified address. It doesn't appear to actually verify that the slave is responding.

THARTARUS, 2019-09-06:
The problem is with Optiboot?

SpenceKonde, 2019-09-06:
No, the problem is with the software i2c implementation, because those
parts don't have hardware i2c master or a USI.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2019-09-07:
Issue is, I think, either here: https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/Wire/src/SoftWire.h or https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/Wire/src/SoftI2CMaster.h

THARTARUS, 2019-09-08:
Now i see the problem, but i need to use an Attiny828 as slave and use an Arduino as master. 
The library on the follow link work?
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/Wire/src/USI_TWI_Slave/USI_TWI_Slave.c

SpenceKonde, 2019-09-08:
OH! You're using it as a slave? This issue only concerns using the 841/441/828 as an I2C master. Slave mode should be fine. 

For all I2C functions with any part supported by ATTinyCore, you should always use <Wire.h>, like you normally would - that version of Wire.h correctly selects the appropriate implementation of the desired functionality on that part. (this issue concerns the implementation of master mode on those three parts - and note that they work fine as masters as long as no bus errors occur)

THARTARUS, 2019-09-08:
I tried to use the "Wire" library on my attiny828(NoBootloader) as slave and my arduino uno (as master) but the master can't find the slave.
I tryed to upload a costum blink code on my attiny828 and it works fine (the Attiny is ok).
I tied to verify the codes (they are the examples from "Wire" library) and i check the connection but both are ok .

Maybe is the attiny configuration that create this problem?

[image]

SpenceKonde, 2020-04-08:
As an aside, to others who stumble across this - I;m sure you already either noticed this or abandoned this project - for the 828, it's in this file https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/Wire/src/WireS.cpp

I'll add a note to the Wire docs describing this a bit better...

The original issue is still present - I have done nothing about it.

SpenceKonde, 2022-02-09:
Not jumping into this shark pit.

closed 2022-02-09 by SpenceKonde