SpenceKonde/ATTinyCore #226: Universal Wire library is space hog on Tiny841/828

SpenceKonde, 2018-07-15:
The Universal Wire library, as a consequence of using two wire implementations (WireS for slave and Sofware I2C Master for master) takes up considerably more space that SofTI2CMaster library did. This is problematic on devices with this little flash; the current flash usage of this library is not acceptable.

SpenceKonde, 2018-11-29:
Can anyone suggest a better solution than an extra menu option for these (Wire support -> Master Only, Slave Only, Both)?

per1234, 2018-11-29:
You could move the relevant code to header files. Then the user can configure it from the sketch:
```c++
#define WIRE_MASTER_ONLY
#include <Wire.h>
```
In the library:
```c++
#ifndef WIRE_MASTER_ONLY
// master code here
#endif 
```
It's not a great solution because often Wire is not used directly in a sketch, but rather by another library which will likely not do this configuration of the Wire library.

SpenceKonde, 2018-12-02:
Yeah - I went with the menu approach. Doesn't save as much as Id hoped for master, but saves significant amount of flash for slave. And ram for both of them because it eliminates a buffer. Still trying to figure out whether there's some code that shouldn't be getting pulled in that is here....

SpenceKonde, 2018-12-02:
Examination of dissassembly results shows that the slave Wire is still being included in the binary, so this change is not complete. Thanks to @per1234 for the assembly listing functionality.

commit 5a868880 (2018-12-02): Fix #226 Fix #267

Also fix issue with class name for I2C master on x41, 828, determine
that TWSWire.* are no longer needed (were only needed because of an
incorect #include in wire.cpp
commit 4810be35 (2018-12-02): Actually fix #226

This change *actually* fixes the inclusion of the slave wire code in the
binary!
closed 2018-12-02 by SpenceKonde