SpenceKonde/ATTinyCore #229: ATTiny85 Wire.endTransmission() hangs

jeremyhahn, 2018-08-01:
I've got 4.7k pull-ups on both SDA and SCL lines between my Mega and tiny. When I remove the SDA wire the loop continues on. I've confirmed the same sketch / wiring works between two megas. 

Are there any known issues that would prevent this from working? What's the best way to debug? Unfortuantely, I don't have access to a scope right now.

SpenceKonde, 2018-08-01:
https://github.com/SpenceKonde/ATTinyCore/issues/208

It is fixed in the github version, but the fix has not yet made it into a board manager release (I had hoped to finish a bunch of things for the board manager release, but have been extremely busy and havent had time to work on the rest of the stuff that was slated for that release). If you remove the board manager installation and do manual install from github it should work: https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md )

jeremyhahn, 2018-08-02:
I removed the old version from boards manager, removed the url from preferences, cloned master, restarted the IDE, ensured the new library is used during compilation - still not working.

SpenceKonde, 2018-08-02:
Please provide code that reproduces the issue

jeremyhahn, 2018-08-02:
// mega master

```
#include <Wire.h>

void setup() {
  Wire.begin();
  Serial.begin(9600);
  Serial1.begin(9600);
}

byte x = 0;

void loop() {
  Serial.println("Starting loop");

  while(Serial1.available()) {
    byte inByte = Serial1.read();
    Serial.write(inByte);
  }

  Serial.println("done checking serial1");

  Wire.beginTransmission(8); // transmit to device #8
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte

  Serial.println("ending transmission");
  
  Wire.endTransmission();    // stop transmitting

  Serial.println("requesting data");

  Wire.requestFrom(8, 6);

  x++;
  delay(500);
}
```

// tiny slave (tried both 1 MHz and 8 MHz internal clocks)

```
#include <Wire.h>
#include <SoftwareSerial.h>

#define rxPin 3
#define txPin 4

SoftwareSerial sserial(rxPin, txPin);

int LED = 0;

void setup() {
  sserial.begin(9600);
  sserial.println("attiny85 initializing");
  
  Wire.begin(8);
  Wire.onReceive(receiveEvent);

  sserial.println("attiny85 initialized");
}

void loop() {
  sserial.println("attiny85 looping");
  delay(100);
}

void receiveEvent(int howMany) {
  while (1 < Wire.available()) {
    char c = Wire.read();
    sserial.println(c);
  }
  int x = Wire.read();
  sserial.println(x);

  delay(1000);
}
```

SpenceKonde, 2018-08-03:
Hmm. 

When you tested it on two megas, were you using software serial or hardware serial? I think your problem is that you're writing to a software serial implementation from within an ISR - unlike hardware serial (which you still should not use within an ISR* but which you can get away with as long as you never fill the buffer such that Serial.print() starts blocking), software serial printing is always blocking.  

Try commenting those lines out and see if it fixes the problem - if not, I'll try to set it up and take a look at it this weekend. 

*(yes, I know the official slaveReceive example does this; the official examples on the Arduino site don't seem to have been vetted/edited by anyone who really knows what they're doing. For example, the standalone breadboard arduino example, and at least several shift register examples omit the 0.1uF decoupling caps, without which the chip may reset or hang intermittently... )

jeremyhahn, 2018-08-03:
Hardware serial was used on the megas, but only local Serial output as there is no need to send the messages anywhere in contrast to the tiny.

I reduced the code to the following, still not working.

// mega master
```
#include <Wire.h>

void setup() {
  Wire.begin();
  Serial.begin(9600);
}

byte x = 0;

void loop() {
  Serial.println("beginTransmission");

  Wire.beginTransmission(8); // transmit to device #8

  Serial.println("writing");
  
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte

  Serial.println("endTransmission");
  
  Wire.endTransmission();    // stop transmitting

  Serial.println("requestFrom");

  Wire.requestFrom(8, 6);

  x++;
  delay(500);
}
```

// tiny slave
```
#include <Wire.h>

int LED = 0;

void setup() {
  Wire.begin(8);
  Wire.onReceive(receiveEvent);
}

void loop() {
  delay(100);
}

void receiveEvent(int howMany) {
  while (1 < Wire.available()) {
    char c = Wire.read();
    Wire.write(c);
  }
  int x = Wire.read();
  Wire.write(x);

  delay(1000);
}
```

Master serial output is:
```
beginTransmission
writing
endTransmission
```

Thanks for looking into it, greatly appreciated!

SpenceKonde, 2018-08-04:
Not quite - remove the wire.write calls. That is the onReceive handler, not the onRequest handler; you should not be writing anything there. For a test that echos what was received, stuff the characters into a buffer and set a flag that you check in loop, and print the buffer, clear it, and unset the flag. Something like that. 

For a slave that the master can send to and request data from, you need two ISRs, not one: 
onReceive for data received from the master, and onRequest for when the master requests data.

jeremyhahn, 2018-08-05:
Thanks, that makes sense. I've refactored to the following, but still hanging on endTransmission with the following code:

// mega master

```
#include <Wire.h>
#include <SoftwareSerial.h>

int x = 1;

void setup() {
  Wire.begin();
  Serial.begin(9600);
  Serial1.begin(9600);
}

void loop() {

  Serial.println("Starting loop");

  while(Serial1.available()) {
    Serial.write(Serial1.read());
  }

  Serial.println("beginTransmission");
  Wire.beginTransmission(8);

  Serial.println("writing");
  Wire.write(x);

  Serial.println("endTransmission"); 
  Wire.endTransmission();

  Serial.println("requestFrom");
  Wire.requestFrom(8, 1);

  while(Wire.available()) {
    unsigned long response = Wire.read();
    Serial.print("Tiny response: ");
    Serial.println(response);
  }
  
  x++;
  delay(500);
}
```

// tiny slave

```
#include <Wire.h>
#include <SoftwareSerial.h>

SoftwareSerial sserial(3, 4);

int LED = 0;

volatile int receiveBytes, receiveCount = 0;
unsigned long lastEvent = 0;

void setup() {
  digitalWrite(LED, LOW);
  Wire.begin(8);
  Wire.onReceive(receiveEvent);
  Wire.onRequest(requestEvent);

  sserial.begin(9600);
}

void loop() {

  sserial.println("ATTiny85 looping!");

  digitalWrite(LED, LOW);
  char msg[75];

  if(receiveBytes > 0) {
    digitalWrite(LED, HIGH);

    sprintf(msg, "ATTiny85 received %i byte message with count %i", receiveBytes, receiveCount);
    sserial.println(msg);

    receiveBytes = 0;
    lastEvent = millis();
  }

  delay(1000);
}

void requestEvent() {
  Wire.write(lastEvent);
}

void receiveEvent(int howMany) {
  receiveBytes = howMany;
  receiveCount = Wire.read();
}
```

If I comment out the begin/endTransmission like this, then it hangs on "requestFrom":

```void loop() {

  Serial.println("Starting loop");

  while(Serial1.available()) {
    Serial.write(Serial1.read());
  }

/*
  Serial.println("beginTransmission");
  Wire.beginTransmission(8);

  Serial.println("writing");
  Wire.write(x);

  Serial.println("endTransmission"); 
  Wire.endTransmission();
*/

  Serial.println("requestFrom");
  Wire.requestFrom(8, 1);

  while(Wire.available()) {
    unsigned long response = Wire.read();
    Serial.print("Tiny response: ");
    Serial.println(response);
  }
  
  x++;
  delay(500);
}
```

If I comment out all of the Wire calls on the master, I get "ATTiny85 looping" serial data from the tiny.

jeremyhahn, 2018-08-14:
I was able to hook up a Nano and remove a couple of bugs, this is what I have now, which works mega <-> mega and mega <-> nano. Still unable to get I2C working with the Tiny, it just hangs.

// mega master
```
#include <Wire.h>

#define SLAVE 0x08

int x = 1;

typedef union {
    char Bytes[4];
    unsigned long Millis;
} LastEvent;

LastEvent lastEvent;

void setup() {
  Wire.begin();
  Serial.begin(9600);
  Serial1.begin(9600);
}

void loop() {

  Serial.println("Starting loop");

  while(Serial1.available()) {
    Serial.write(Serial1.read());
  }
  Serial.println();

  Serial.println("beginTransmission");
  Wire.beginTransmission(SLAVE);

  Serial.println("writing");
  Wire.write(x);

  Serial.println("endTransmission"); 
  Wire.endTransmission();

  Serial.println("requestFrom");
  Wire.requestFrom(SLAVE, sizeof(lastEvent.Bytes));

  int numBytes = Wire.available();
  for(int i=0; i<numBytes; i++) {
    lastEvent.Bytes[i] = Wire.read();
  }
  Serial.print("Tiny response: ");
  Serial.println(lastEvent.Millis);

  Serial.println();

  x++;
  delay(500);
}
```

// nano / tiny slave
```
#include <Wire.h>

#define MYADDRESS 0x08

int LED = 0;

typedef union {
    char Bytes[4];
    unsigned long Millis;
} LastEvent;

typedef union {
    char Bytes[4];
    int Count;
} Message;

LastEvent lastEvent;
volatile int receiveBytes;
volatile Message message;

void setup() {

  digitalWrite(LED, LOW);
  
  Wire.begin(MYADDRESS);
  Wire.onReceive(receiveEvent);
  Wire.onRequest(requestEvent);

  Serial.begin(9600);
  Serial.println("Nano initialized");
}

void loop() {

  Serial.println("Nano looping!");

  digitalWrite(LED, LOW);

  if(receiveBytes > 0) {
    digitalWrite(LED, HIGH);

    Serial.print("Received bytes: ");
    Serial.println(receiveBytes);

    Serial.print("message.Count: ");
    Serial.println(message.Count);

    lastEvent.Millis = millis();
    receiveBytes = 0;
  }

  delay(250);
}

void requestEvent() {
  Wire.write(lastEvent.Bytes, sizeof(lastEvent.Bytes));
}

void receiveEvent(int howMany) {
  receiveBytes = howMany;
  for(int i=0; i<receiveBytes; i++) {
    message.Bytes[i] = Wire.read();
  }
}
```

I'm going to use the Nano to unblock myself for now but it would sure be nice to have the smaller footprint of the tiny.

SpenceKonde, 2022-01-26:
Never reproduced.

closed 2022-01-26 by SpenceKonde