SpenceKonde/ATTinyCore #230: I2C and serial monitor breaking the i2c connection

battlewii, 2018-08-03:
i used the Wire examples from ATtinycore, ATtiny85 as master_writer and canique mk2 with an fdti adapter (bub iii) as slave_receiver.
The second i open the serial monitor the rx led on the adapter stops, when i then let the serial monitor open and reset the Attiny it works fine.
I tested the same examples with an Arduino Uno and the canique mk2 + adapter works fine when open the serial monitor. I have already cloned the github version but nothing has changed.

SpenceKonde, 2018-08-04:
See also #229 - seems this is "exerise USI I2C slave" week. 

I have not vetted the examples. I realized while investigating #229 that the official arduino library examples do things like writing to serial within the ISR, which is bad practice with hardware serial, but an absolute no-go with sofware serial like is the case on most parts with an I2C.

SpenceKonde, 2018-08-04:
Anyway though, thinking about this, I feel like this is a corner case - if you reset the master during an I2C operation, that may well not be handled correctly. Thought is required to determine how to handle this gracefully

Can you post the two programs you're using? 

The Canique Mk2 looks like essentially a pro mini (which is what I have to test like), so a pair of sketches that would work with a tiny85 and a pro mini would be ideal.

SpenceKonde, 2018-08-13:
Can you help out by posting code that demonstrates the problem?

battlewii, 2018-08-14:
sry didn't had much time to post the sketches, but well i was just using the default example sketches.
Master:
`// Wire Master Writer
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Writes data to an I2C/TWI slave device
// Refer to the "Wire Slave Receiver" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup() {
  Wire.begin(); // join i2c bus (address optional for master)
}

byte x = 0;

void loop() {
  Wire.beginTransmission(8); // transmit to device #8
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte
  Wire.endTransmission();    // stop transmitting

  x++;
  delay(500);
}`

Slave:
`// Wire Slave Receiver
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Receives data as an I2C/TWI slave device
// Refer to the "Wire Master Writer" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup() {
  Wire.begin(8);                // join i2c bus with address #8
  Wire.onReceive(receiveEvent); // register event
  Serial.begin(9600);           // start serial for output
}

void loop() {
  delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
  while (1 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  int x = Wire.read();    // receive byte as an integer
  Serial.println(x);         // print the integer
}`

SpenceKonde, 2018-11-29:
I re-read this a few times while looking through issues

So what happens is you're resetting the slave, but the master stops writing at that point? 
I think the issue is that the master isn't ever timing out the failed transactions (which occur while the bootloader is running or if the slave is reset during the transaction)

SpenceKonde, 2020-10-03:
Has been marked mystery for over a year, in absence of additional information from users experiencing this issue. Closing.

closed 2020-10-03 by SpenceKonde