SpenceKonde/ATTinyCore #231: {Support|Example} for ATTiny85 SPI as Slave is {missing|wished} (?)

sjjh, 2018-08-03:
I'm currently trying to bend my head around programming an ATTiny85 with an Arduino Uno to be used as a SPI slave. Because of the SPI library I chose this project. After searching on the net and going to the files available here (especially at ATTinyCore/avr/libraries/SPI/examples/) I'm not sure if there's actually SPI slave support for the ATTiny85 by the library and if so, how to use it.
I thus would love to get some clarity
1) if SPI slave mode for the ATTiny85 is supported by ATTinyCore
2) how it actually works (some examples would be great)

If the answer to 1) is "no", any idea for another library to prefer, or isn't it too hard to write the basic communication code by yourself?
Any help appreciated! Thx in advance

SpenceKonde, 2018-08-04:
No, slave SPI is not supported by the universal SPI library. 

This library reimplements the functions of the Arduino SPI library for USI, and uses that on parts with a USI instead of SPI. Slave SPI is not supported by the official SPI library, hence it is not supported by this one. 

I have not studied the available SPI slave library implementations. If there's a standard SPI slave library, and a USI SPI slave library that implements the same interface, they could be made into a universal SPI slave library (named the same thing as the popularly used SPI slave library). But I am not aware of either of the two libraries that would need to go into this!

closed 2018-08-04 by SpenceKonde

sjjh, 2018-08-04:
Thanks for the quick feedback.