SpenceKonde/ATTinyCore #234: Write.begin() hang

martinpcbox, 2018-08-11:
Hi.

I want to add 16x2 I2C LCD display to atTiny85. The display work well with Arduino Mega. So I try to connect the display to atTiny with the same sketch but it do not work. After some test I found than Wire.begin hang.

  
// on first test, LED = 100 and come on. on second test LED=0; 
  _analogWrite(LED_PIN,LED);_ 
  _lcd.init();                      // initialize the lcd_ 
// This time Led does not show. So I suspect there is something wrong in lcd.init(). I cheched the code:
_// void LiquidCrystal_I2C::init_priv()_
_//{_
_//	Wire.begin();_
_//	_displayfunction = LCD_4BITMODE | LCD_1LINE | LCD_5x8DOTS;_
_//	begin(_cols, _rows);_  
_//}_
// Very simple so only Write.begin() should be the problem.
  _analogWrite(LED_PIN, 100);_

Is there a fix to this problem?

SpenceKonde, 2018-08-13:
I have done things almost identical to this without problem (not on the tiny85, but on boards with USI for I2C master, I will try to find a chance to investigate this, however I am extremely busy currently and can provide no estimate on when I would be able to take a look. 

It should just work. Are you sure the it is connected correctly and that there are pullups on SCL and SDA? Have you triple checked that all connections are complete.

SpenceKonde, 2018-08-17:
Can you please post a complete sketch that demonstrates this problem? 

I have plenty of 16x2 I2C LCDs to test with.