SpenceKonde/ATTinyCore #236: Add support for ATTiny26

Master811129, 2018-08-14:
the attiny26 /v is a useful MCU in the TinyAVR  family.
plzz add it
thanks

SpenceKonde, 2018-08-14:
This is not a priority - the core already supports the strictly better tiny261, 461, and 861 and those are readily available with a much better feature set, and as there's no larger version, you're stuck with 2k of flash, which is really constraining due to the overhead of arduino functionality like timekeeping and pin numbers. 

If you add support, I'll roll it into the core (what needs to be done: Small modifications to pins_arduino.h, and a new #ifdef block in the functions that manipulate the PWM timer 1) - right now I am absolutely swamped with work and projects, and cannot give any estimate of when I would be able to implement support myself. 

I wonder if this might be a better fit for @MCUdude MicroCore which is already optimized for the extreme low flash ATTiny13 - he's done a lot of work to optimize his core for low flash parts like this.

MCUdude, 2018-08-14:
I don't feel like adding a "rarely used" microcontroller such as ATtiny26 to MicroCore. MicroCore is highly optimized for ATtiny13, and will require a lot of work to add other microcontrollers. The plan is to add other <=1kB devices such as ATtiny102 and Attiny104 in the future but I haven't really bothered yet.

I'm sure it's a pretty straight forward job adding it to ATtinyCore. However 2kB is pretty limited if the code isn't tightly optimized for the hardware.

sleemanj, 2018-08-14:
In my fork of ATTinyCore which is better optimised for 13 (and 4/5/9/10 although usefulness of those is more limited than 13 due to the reduced instruction set) I have had it listed in my [FUTURE.md](https://github.com/sleemanj/optiboot/blob/master/dists/FUTURE.md) for some time and my notes at the time were it was less likely for me to get around to including it in my fork since it's broadly similar to 87/167 with port name changes.

In other words, I figured that it probably just isn't used enough to justify effort.

SpenceKonde, 2018-11-12:
Closed. No plans to change. If someone wants to add it and send a pull request, I'm happy to merge it, but otherwise, I have no plans to add it.

closed 2018-11-12 by SpenceKonde