SpenceKonde/ATTinyCore #237: ATTiny85 SDA line stuck low

giotherobot, 2018-08-17:
i'm using this core with platformIo to program an ATTiny85 to talk to an ESP8266 via I2C. 
The job of the ATTiny is to dim an incadescent light, so i'm using timer1 and pin interrupts.
The problem is that after a couple seconds of good communication the ATtiny keeps the sda line low.
I traced the problem to be some sort of incompatibility with timer1 interrupts, from the avr libc manual i set all my interrupts to non-blocking. This last attempt seemed to solve the problem at first but it represented itself.
Any idea how to solve?
Thanks!

SpenceKonde, 2018-08-17:
Are you using the latest version of the core (from github - not board manager)? There is a fix for a problem with that symptom that is in the github version, but has not yet made it into the board manager release. 

There appear to be several issues relating to I2C on chips using a USI that I have not yet had time to investigate.

SpenceKonde, 2018-08-17:
The issue in question is #208

giotherobot, 2018-08-17:
No i was using the version from board manager, will try now with git version

giotherobot, 2018-08-17:
You just saved my thesis :+1: 
Problem seems to be solved with last git version!
Fantastic!!
Thanks a lot!

SpenceKonde, 2018-08-17:
Great. I hope to get to take a look at the other USI I2C related issues people have posted up and get a new release out to bring in these fixes.

giotherobot, 2018-08-17:
I sadly kept trying, and the issue is still present :(

Now, i have just put the git cloned folder in the lib folder of my project, but i'm not sure if platformio is using that or the board manager one. 

If there is any test i can do to try and solve this i will, i kinda need this...

SpenceKonde, 2018-08-20:
Hm, I am not familiar with platformio. If you can get the verbose compiler output, you can see which files it's using... if it's using the git version of the library, this is troubling indeed. 

What I recommend is uninstalling the board manager version entirely, and doing a manual install of the git version (I clone it into (sketchbook)/hardware ).

giotherobot, 2018-08-20:
I need to keep working on my thesis, so i switched to an ATMega. I will get back to testing this when i'm finished. I like the ATTinys and would love to use them again, so I'll be back ;)

SpenceKonde, 2018-12-02:
Reread above. 

Closing, this is #208, and per aug17 comment it appears that the modified library wasn't installed in a way that it was used.

SpenceKonde, 2018-12-02:
Reopen if this turns out not to be the case. Any recent release should have this fixed.

closed 2018-12-02 by SpenceKonde