SpenceKonde/ATTinyCore #238: what boards should have TwoWire class support?

tedder, 2018-08-23:
Spence, I'm down a rabbit hole. I can import Wire.h, but an upstream library thinks it should have TwoWire. Based on the errors, I'm getting TWSWire.

The upstream code tries this:

    TwoWire *_i2cPort;
    boolean begin(TwoWire &wirePort = Wire);

and that fails with the following:

    error: 'TwoWire' does not name a type
    error: 'TwoWire' is not a type
    error: could not convert 'Wire' from 'TWSWire' to 'int&' 

What's the answer? Should the upstream be using Wire? Should TWSWire masquerade as TwoWire to match the stock Arduino's Wire class?

SpenceKonde, 2018-08-23:
Yes, TWSWire class should be named TwoWire (and #defines should be sufficient to keep any conflicts away).

tedder, 2018-08-23:
Hmmm. I had tried a bunch of things but touched 'your' Wire.h last. By replacing all the class names, but leaving the `#define`s and filenames alone, it worked. Want a PR for it? You might have a smarter way to do it.

SpenceKonde, 2018-08-23:
By all means, send over a PR!! <3

closed 2018-08-24 by SpenceKonde