SpenceKonde/ATTinyCore #240: delete_merged_output.sh: No such file or directory

toothrobber, 2018-08-28:
I am trying too export binary (attend-blink) in the Arduino IDE.

can someone point me to a copy of delete_merged_output.sh

Robert

/Users/robertabooth/Library/Arduino15/packages/diy_attiny/hardware/avr/2018.3.11/delete_merged_output.sh 
chmod: /Users/robertabooth/Library/Arduino15/packages/diy_attiny/hardware/avr/2018.3.11/delete_merged_output.sh: No such file or directory
Error compiling.

per1234, 2018-08-28:
You're using [diy_attiny](https://github.com/sleemanj/optiboot), not ATTinyCore, so you posted this to the wrong repository.

SpenceKonde, 2018-08-28:
Indeed. My core includes this file! My guess would be that they took my platform.txt and used it without realizing they also needed to gank the delete_merged_output scripts.

closed 2018-08-28 by SpenceKonde

sleemanj, 2018-08-28:
Fwiw, while the merged output isn't something I support particularly,
those missing scripts where added back in March.  So ticket poster you
might was to update your version selected in boards manager, if it still
is an issue open a ticket in the repo linked by @per1234 and I'll take a
look.