SpenceKonde/ATTinyCore #241: ATTINY 861 WDT

hja685, 2018-09-08:
I am writing because I am using your board layout in a project with a ATTINY 861. I am attempting to use a watchdog timer to wake from "SLEEP_MODE_PWR_DOWN."

I need this to work in order to be able to optimize my power consumption before rolling this project out. 

I am attempting to modify the WDTCR, and after reading your readme, it appears that this may not be possible due to bootloader constraints. 

Is it impossible to get this to work? Or will I need to rewrite the bootloader structure for this to work? 

I am willing to write this myself, just trying to understand why I am getting an error that WDTCR is not defined when attempting to make changes to this register. 

-hja685

hja685, 2018-09-08:
I'm dumb. Issue resolved on my own. I was using WDTCSR rather than WDTCR. 

It compiles, so I'm assuming it should still work?

SpenceKonde, 2018-09-08:
Yes, it should work without issue

It's only a problem with boards that have a bootloader, and only when the
bootloader is in use. We dont have an 861 bootloader, and even if we did,
you could disregard this issue by just using ISP to program

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

sleemanj, 2018-09-08:
@SpenceKonde you should change those virtual boot bootloaders to use
EE_RDY instead of the WDT vector, EE_RDY is very rarely used especially
in arduino land, eeprom library doesn't use it.  wdt by comparison is
used often 

 

Links:
------
[1]
https://github.com/SpenceKonde/ATTinyCore/issues/241#issuecomment-419627496
[2]
https://github.com/