SpenceKonde/ATTinyCore #245: Internal temperature sensor - ATTINYx4

ludek111, 2018-09-18:
Hi,
I was trying to read temperature of internal temp sensor.
>  AnalogRead(8)

 doesn't work. Manually setting registers also doesn't work. For example:
  
> void setup(){
>   analogReference(INTERNAL);
>   ADMUX |= (0 << MUX4);  //Temperature Sensor - 100111
>   ADMUX |= (0 << MUX3);  //Temperature Sensor - 100111
>   ADMUX |= (1 << MUX2);  //Temperature Sensor - 100111
>   ADMUX |= (1 << MUX1);  //Temperature Sensor - 100111
>   ADMUX |= (1 << MUX0);  //Temperature Sensor - 100111
> 
>   //ADC Control and Status Register A 
>   ADCSRA = 0;
>   ADCSRA |= (1 << ADEN);  //Enable the ADC
>   ADCSRA |= (1 << ADPS2);  //ADC Prescaler - 16 (16MHz -> 1MHz)
> 
>   //ADC Control and Status Register B 
>   ADCSRB = 0;
>   ADCSRB |= (1 << MUX5);  //Temperature Sensor - 100111
>   mySerial.begin(9600);
> }
> 
> void loop() {
>   ADCSRA |= (1 << ADSC);  //Start temperature conversion
>   while (bit_is_set(ADCSRA, ADSC));  //Wait for conversion to finish
>   byte low  = ADCL;
>   byte high = ADCH;
>   int temperature = (high << 8) | low;  //Result is in kelvin
>   mySerial.print("ADC: ");
>   mySerial.println(temperature);
>   delay(500);
> }

Second thing I want to say is that
 >   analogReference(INTERNAL);

works perfectly, (compared with my zotek multimeter, internal reference is ~1.09V).... but for example:

> ADMUX = _BV(REFS1) | _BV(REFS0);

doesn't work. I also tried code from there:

> https://playground.arduino.cc/Main/InternalTemperatureSensor

I am using Attiny44A and I know it's not counterfit.

sleemanj, 2018-09-18:
> AnalogRead(8)

You should describe **what** doesn't work.  Do you get a compile error, is it giving wrong values, does it do nothing....

Note the datasheet page 128 

> The internal 1.1V reference must also be selected for the ADC reference source in the temperature sensor measurement. 

So be certain you selected the internal reference before taking a reading of the sensor.

Additionally, the datasheet page 130 indicates the MUX value for ADC8 is actually 100010 which is 34 decimal, I'd have to look at the variant code to be certain but you probably need to do `analogRead(34)` not `analogRead(8)`.

> ADMUX |= (0 << MUX4); //Temperature Sensor - 100111
> ADMUX |= (0 << MUX3); //Temperature Sensor - 100111
> ADMUX |= (1 << MUX2); //Temperature Sensor - 100111
> ADMUX |= (1 << MUX1); //Temperature Sensor - 100111
> ADMUX |= (1 << MUX0); //Temperature Sensor - 100111

As above, page 130 (and 128) of the datasheet indicates that to select ADC8 you should utilise the bit pattern for bits 5:0 of ADCMUX to 100010 - the selection you have made above appears incorrect.

> ADMUX = _BV(REFS1) | _BV(REFS0);

See datasheet page 129 for internal reference selection, your selection of REFs appears to be incorrect.

ludek111, 2018-09-18:
I figured out how to get +/- good readings! But If I compile:

< AnalogRead(8)

what I get is somewhat like random values... (for example: ADCVAL=0, 0, 24, 0, 3, 743, 0, 0, 74, 0, 355, 0, 2, 0...)
Voltage reference: This works fine now: (setting internal 1.1V reference)

>   ADMUX |= (1 << REFS1);
>   ADMUX |= (0 << REFS0);

 this also works good:

>analogReference(INTERNAL);

**### below "finally I figured out!" section:**

I don't know why this works now, but:
I couldn't get good temperature values (they were completely different if I change my Attiny voltage from 3v3 to 5.0V), but now, my code is like this and it works:

`#include <SoftwareSerial.h>
int sensorValue = 0;
const int sensorI = 2;
SoftwareSerial mySerial(7, 8); // RX, TX

void setup(){
  analogReference(INTERNAL);
  ADMUX |= (1 << REFS1);
  ADMUX |= (0 << REFS0);
  ADMUX |= (1 << MUX5);
  ADMUX |= (0 << MUX4);  //Temperature Sensor - 100010 
  ADMUX |= (0 << MUX3);  //Temperature Sensor - 100010 
  ADMUX |= (0 << MUX2);  //Temperature Sensor - 100010 
  ADMUX |= (1 << MUX1);  //Temperature Sensor - 100010 
  ADMUX |= (0 << MUX0);  //Temperature Sensor - 100010 

  ADCSRA = 0;
  ADCSRA |= (1 << ADEN);  //Enable the ADC
  ADCSRA |= (1 << ADPS2);  //ADC Prescaler - 16 (16MHz -> 1MHz)

//  ADCSRB = 0;

  mySerial.begin(9600);
}

void loop() {
  ADCSRA |= (1 << ADSC);  //Start temperature conversion
  while (bit_is_set(ADCSRA, ADSC));  //Wait for conversion to finish
  byte low  = ADCL;
  byte high = ADCH;
  int temperature = (high << 8) | low;  //Result is in kelvin
  mySerial.print("ADC: ");
  mySerial.println(temperature);
//  mySerial.print(" temp val ");
//  ADMUX = bit (REFS0) | bit (REFS1)  | 0x08; 
//  bitSet (ADCSRA, ADSC);  // start a conversion  
//  while (bit_is_set(ADCSRA, ADSC))
//    { }
//  mySerial.println(ADC);
  
  delay(500);
}`

I got some values AND I plotted this. First is compressed air (< -20*C) and the function rise is 60W "classic" light bulb.

[https://abload.de/img/clipboard01w5ds0.png](url)

So, everything works, but not as elegant as AnalogRead(8);
thanks!

ludek111, 2018-09-19:
Thanks alot! So this works perfectly:

```
#include <SoftwareSerial.h>
SoftwareSerial mySerial(7, 8); // RX, TX

void setup(){
  analogReference(INTERNAL);
  mySerial.begin(9600);
}

void loop() {
  int temperature=analogRead(34);
  mySerial.print("ADC: ");
  mySerial.println(temperature);
  delay(500);

```
learning arduino coding on attiny84 (non-default chip) is harder, but more satisfying :)

SpenceKonde, 2018-09-19:
Thanks for getting this sorted out!

SpenceKonde, 2018-10-08:
This should now work on all parts. 

(note, using Serial - all parts have either a hardware serial port, or a builtin software serial port on the AIN0/AIN1 pins (the builtin software serial trades flexibility about pin assignment for not using a PCINT, improving compatibility with sketches and libraries)

```

void setup(){
  analogReference(INTERNAL); //You must use the internal 1.1v bandgap reference when measuring temperature
  Serial.begin(9600);
}

void loop() {
  int temperature=analogRead(ADC_TEMPERATURE); //ADC_TEMPERATURE is #defined to be the channel for reading the temperature; this varies between parts. 
  Serial.print("ADC: ");
  Serial.println(temperature);
  delay(500);
}

```

SpenceKonde, 2018-10-08:
Will be rolled into core (hopefully as an example, potentially as a .md (but nobody reads those anyway :-( )) pending an answer here https://forum.arduino.cc/index.php?topic=572440

commit 7c5e4882 (2018-10-09): Add temp read example - fix #245
closed 2018-10-09 by SpenceKonde