SpenceKonde/ATTinyCore #250: How to fuse the ATTiny85 Reset Pin into I/O Pin ?

nagaokashi, 2018-09-23:
I understood that once the Reset Pin has been disabled, SPI can not be used anymore.
Event this behavior, I really want to change Reset Pin of ATTiny85 to a normal I/O pin.
But I don't know how to do it.
(I researched on google but did not find the answer which I could understand)
Maybe I misunderstand, I think the progress is as below:
Step 1. Do a awesome program 
Step 2. Upload to ATTiny85 by using Arduino with SPI
Step 3. Fuse Attiny85 to disable Reset Pin
Q: How to do step 3 ? 

Thanks.

sleemanj, 2018-09-23:
Use avrdude (with your ISP programmer) to set the fuses appropriately, example

    avrdude -c usbasp -Pusb  -U lfuse:w:0xE2:m	-U hfuse:w:0x7D:m	-U efuse:w:0xFF:m	-U lock:w:0xFF:m

You can use burn bootloader and observe the console window to get the command line for running your IDE's copy of avrdude.

Use a fuse calculator to determine the appropriate fuses, such as this one

  http://eleccelerator.com/fusecalc/fusecalc.php?chip=attiny85

you can see the avrdude options for fuse under the fuse selections

sleemanj, 2018-09-23:
And remember, once you disable reset there is no going back, you will not be able to program the chip again or re-enable reset unless you get a HVPP or "Fuse Doctor" device.

nagaokashi, 2018-09-24:
@sleemanj  Thank you very much. I'll try and feedback.
About HVP, I also googled this schematic, I think the based-Arduino HVP is quite easy to create. That is motivation for me to decided to disable Reset Pin.

SpenceKonde, 2018-09-24:
Note - it's HVSP (high voltage serial programming) not HVPP (high voltage parallel programming) for the attiny85 and most of the ATTiny series. HVPP is used on atmega chips and maybe some tinies with lots of pins - most tinies don't have enough pins for parallel programming. 

For the fuses - I would instead of using one of those fuse calcs, copy the first avrdude invocation from when I did "burn bootloader" to set up the device, and just modify the RSTDSBL bit. That way, you know you're not changing some other fuse you didn't intend to. And if I needed to reset the fuses to reprogram, I would not try to make the computer talk to it via HVSP - I'd make a fuse doctor instead. 

Be sure you've got all the fuses the way you like them, especially if you're blowing the lock fuses too to prevent reading of the flash/eeprom. I recently talked to someone who very nearly got himself up shit's creek without a paddle, with hundreds of devices that needed the eeprom preserved but with the fuses not set to do that and the lockbits blown; luckily a careful analysis of his requirements let to his realization that he didn't actually need to do that. But I could definitely sense the moment of panic. In a similar way, you can paint yourself into a corner with RSTDSBL as well. 

Also - be sure to review the electrical properties section of the ATTiny85 datasheet, specifically the output driver and pullup strength of the reset pin when used as I/O - it's much weaker than other pins!

nagaokashi, 2018-09-24:
Thank you very much for your detail explanation, @SpenceKonde 

> I'd make a fuse doctor instead, hit it with that, and then reprogram via ISP
Yes, this is what I expected. 

> Also - be sure to review the electrical properties section of the ATTiny85 datasheet, specifically the output driver and pullup strength of the reset pin when used as I/O - it's much weaker than other pins!

I intend to use the Reset pin as an output in order to control a transistor which works as switch for controlling a motor. I think (just think) the output driver of Reset Pin is enough for this purpose.  I tried to confirm via ATTiny85 datasheet  but with my poor knowledge I did not find the answer. So I though , let paint myself. 

As you said, "the eeprom preserved" is my concern. Although I've not tried yet, but as @sleemanj  and you said, I'll follow the "Bun Bootloader"  instruction for reference of avrdude. I just wonder that "the eeprom preserved" option whether is included with the "Bun Bootloader" or not ?

SpenceKonde, 2018-09-24:
When the Attiny85 board is selected (at least using my core) there is a submenu to control whether EESAVE will be set when doing burn bootloader.

nagaokashi, 2018-09-24:
Thank you. I've been using this option from beginning. 
I'd expect this option is also included into "Burn Bootloader"'s avrdude command line. I'll try and feedback.

SpenceKonde, 2018-09-24:
Yeah, it is.

jassing, 2018-09-24:
I may have an aextra fuse doctor You could borrow, trade,rent, or buy....

JohnOH, 2018-09-24:
I am interested in a Fuse Doctor, is it HVPP & HVSP?

nagaokashi, 2018-09-27:
I'd link to feedback that the Fuse DISALBE RESET works fine at the first run. So I have no chance to confirm "Fuse Doctor" yet. I'll try and feedback later.

`avrdude -CC:\Users\xxxx\Documents\Arduino\hardware\ATTinyCore\avr/avrdude.conf -v -pattiny85 -carduino -PCOM4 -b19200 -e -Uefuse:w:0xFF:m -Uhfuse:w:0b**0**1010111:m -Ulfuse:w:0xE2:m -Uflash:w:C:\Users\xxxx\AppData\Local\Temp\arduino_build_924913/ATiny85.ino.hex:i`

closed 2018-09-27 by nagaokashi

SpenceKonde, 2018-09-27:
Thanks for confirming that :-) 

I think I will throw together a .md file with a bit of discussion on this, since I do occasionally get asked about it.

nagaokashi, 2018-09-28:
I just would to feedback that the Fuse Doctor works like charm. :))
Refer from this <a href="https://www.instructables.com/id/Simple-and-cheap-Fuse-Doctor-for-Attiny/">link</a>
And this <a href="https://github.com/nagaokashi/ATTiny_FuseDoctor">link</a> just for above link die.