SpenceKonde/ATTinyCore #255: Unable to burn Bootloader on Mac

coryk88, 2018-10-23:
Now having issues burning bootloader on Mac, able to on W10.

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/corykersey/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.2.0/avrdude.conf"
         User configuration file is "/Users/corykersey/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1431
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny461
         Chip Erase delay              : 15000 us
         PAGEL                         : PB3
         BS2                           : PB2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     4    0 no        256    4     64  4000  4000 0xff 0xff
           flash         65     6    64    0 yes      4096   64     64  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 425479
         Firmware Version: 8450052.-1541889796
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

Error while burning bootloader

SpenceKonde, 2018-10-23:
Closing as this is not related to attinycore. 

0x14 0x02 0x10 is a common error encountered when using Arduino as ISP (it is failing to talk to the arduino being used as programmer - before it even tries to initiate communication with the target), and I have not been able to determine what the cause is (a brief perusal of the forums indicates that this is frequently reported). Some people report fixing it by disabling reset on the board being used as programmer (~10uF cap between reset of the programmer and ground is the easiest way - any cap larger than 10uF will work, within reason). 

I recommend using a real ISP programmer, as there seem to be so many "gotchas" when using the Arduino as ISP - I use the USBAsp and USBTinyISP programmers.

closed 2018-10-23 by SpenceKonde

coryk88, 2018-10-23:
Thanks for the clarification! Will try one of the programmers you mentioned in your comment.