SpenceKonde/ATTinyCore #256: Attiny 1634

robpan3, 2018-10-31:
Hallo,
I'd like to add the board Attiny1634 to my Arduino IDE.
On Internet I couldn't find anything.
Can you help me?
Thanks a lot!
Roberto
Graz
Austria
r.pangallo@gmail.com

SpenceKonde, 2018-10-31:
Follow the instructions here: https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

After following those instructions, the ATTiny1634 (as well as the other many supported ATTtiny devices) will appear in the tools -> board menu. 

Note that this only works with the installed version of the IDE, not the CREATE web IDE, which does not support third party hardware.

SpenceKonde, 2018-10-31:
Closing issues as not a bug, you may continue to comment on the closed issue.

closed 2018-10-31 by SpenceKonde

robpan3, 2018-10-31:
Thanks a lot!I will try.RobertoGrazAustriaInviato dal mio dispositivo mobile Huawei-------- Messaggio originale --------Oggetto: Re: [SpenceKonde/ATTinyCore] Attiny 1634 (#256)Da: "Spence Konde (aka Dr. Azzy)" A: SpenceKonde/ATTinyCore CC: robpan3 ,Author Follow the instructions here: https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md
After following those instructions, the ATTiny1634 (as well as the other many supported ATTtiny devices) will appear in the tools -> board menu.
Note that this only works with the installed version of the IDE, not the CREATE web IDE, which does not support third party hardware.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/SpenceKonde/ATTinyCore/issues/256#issuecomment-434731476",
"url": "https://github.com/SpenceKonde/ATTinyCore/issues/256#issuecomment-434731476",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [SpenceKonde/ATTinyCore] Attiny 1634 (#256)",
"sections": [
{
"text": "",
"activityTitle": "**Spence Konde (aka Dr. Azzy)**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@SpenceKonde",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"SpenceKonde/ATTinyCore\",\n\"issueId\": 256,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/SpenceKonde/ATTinyCore/issues/256#issuecomment-434731476"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 403637467\n}"
}
],
"themeColor": "26292E"
}
]

SpenceKonde, 2018-10-31:
Did you actually install my core using board manager after adding the json URL? 

It sounds like either you only have the leondardomiliani core installed, or you just need to scroll down in the list of boards (if it has been installed, there should be another list of attiny's in tools -> board). My boards are even named differently from his (I combine all the parts that differ only in flash size into a single entry to reduce clutter in the board menu).

robpan3, 2018-10-31:
Thanks (I don't know your name),with your suggestions (delete the leonardomilani core and install your board, everything is good, very good (I can see all the models of AttinyVery very good!Can I know your name and your country?Thanks a lot. You was very kind.Ciao! Please to meet you.Excuse me for my poor English.I am from Italy but I live in Austria with my family, and Electronics + Informatics are my hobby since  40  years.RobertoGrazAustria-------- Messaggio originale --------Oggetto: Re: [SpenceKonde/ATTinyCore] Attiny 1634 (#256)Da: "Spence Konde (aka Dr. Azzy)" A: SpenceKonde/ATTinyCore CC: robpan3 ,Author Did you actually install my core using board manager after adding the json URL?
It sounds like either you only have the leondardomiliani core installed, or you just need to scroll down in the list of boards (if it has been installed, there should be another list of attiny's in tools -> board). My boards are even named differently from his (I combine all the parts that differ only in flash size into a single entry to reduce clutter in the board menu).

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/SpenceKonde/ATTinyCore/issues/256#issuecomment-434804897",
"url": "https://github.com/SpenceKonde/ATTinyCore/issues/256#issuecomment-434804897",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [SpenceKonde/ATTinyCore] Attiny 1634 (#256)",
"sections": [
{
"text": "",
"activityTitle": "**Spence Konde (aka Dr. Azzy)**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@SpenceKonde",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"SpenceKonde/ATTinyCore\",\n\"issueId\": 256,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/SpenceKonde/ATTinyCore/issues/256#issuecomment-434804897"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 403637467\n}"
}
],
"themeColor": "26292E"
}
]

SpenceKonde, 2018-10-31:
I'm Spence Konde, from the USA. Got into the hobby much more recently, and have been turning it into a business, selling ATTiny breakout boards and prototyping board from my Tindie store: https://www.tindie.com/stores/DrAzzy/

-Spence

robpan3, 2018-10-31:
Thanks, Spence.Very good presentation!I love too the Attiny. More than other microcontroller too commercial.The Attiny needs more attention using less memory less pins (sometime) and required more "brain". There is less documentation, but more satisfaction to made itself.I will remember your shop for my needed.Ciao! Roberto-------- Messaggio originale --------Oggetto: Re: [SpenceKonde/ATTinyCore] Attiny 1634 (#256)Da: "Spence Konde (aka Dr. Azzy)" A: SpenceKonde/ATTinyCore CC: robpan3 ,Author I'm Spence Konde, from the USA. Got into the hobby much more recently, and have been turning it into a business, selling ATTiny breakout boards and prototyping board from my Tindie store: https://www.tindie.com/stores/DrAzzy/
-Spence