SpenceKonde/ATTinyCore #257: Some chips in tinymodern do not define NUM_DIGITAL_PINS and etc.

WestfW, 2018-11-07:
It seems to be pretty standard to define NUM_DIGITAL_PINS and similar in the pins_arduino.h file (usually but not always in variant/xxxx)
The cores/tinymodern seems to use cores/pins_arduino.h and cores/core_pins.h instead of the variant mechanism, with several chips defined in each file, but it does NOT define NUM_DIGITAL_PINS (it has CORE_DIGITAL_LAST, which is sort-of the same thing.)
(come to think - I'm not sure that it's clearly defined how exactly the NUM_ macros are supposed to be defined, especially when it comes to overlapping.)

In any case, the test program I wrote that does tries digitalWriteFast() on "all of the digital pins" fails on ATtiny1634, because I'm trying to do 0..NUM_DIGITAL_PINS...

per1234, 2018-11-07:
>I'm not sure that it's clearly defined how exactly the NUM_ macros are supposed to be defined, especially when it comes to overlapping.

It is unfortunate these standardized parts of the variant definitions aren't documented in the Arduino hardware specification. Lacking that, I'd say [arduino:avr:standard](https://github.com/arduino/ArduinoCore-avr/blob/master/variants/standard/pins_arduino.h#L28-L29) should be used as the reference:
```cpp
#define NUM_DIGITAL_PINS            20
#define NUM_ANALOG_INPUTS           6
```
That also seems the most logical way to do it.

I don't understand the strange values of `NUM_ANALOG_INPUTS` in most of the ATTinyCore variants. I don't see that they're used anywhere in the core.

SpenceKonde, 2018-11-07:
Thanks for this report, I'll make sure to get the missing NUM_DIGITAL_PINS defined where missing for the next release. 

The reason for the large values of NUM_ANALOG_INPUTS was to permit the analogRead() function to be used with the differential ADC input channels and temperature sensor. Because in the past there was an explicit test for it. I wound up removing that a while back, but never changed NUM_ANALOG_INPUTS back when I redid that a while back. I'll fix this too, assuming I also can't find anywhere that depends on this. 

I'm not particularly happy with the fact that the variants system isn't used for the tiny modern core. It's something that ought to be fixed (and may need to be sooner rather than later, as I'm supporting the less-stupid pin numbering for the tiny841, to coincide with the redesigned Tiny841 board that I plan to release this winter) - what happened is that the tiny841 and 1634 were supported first, and I followed the example from the original core that I adapted that offered partial support for the tiny841. Back then I was less experienced with Arduino and didn't realize how dumb this was.

SpenceKonde, 2018-11-07:
Also, let me know if you find any other missing #defines

commit d5b67f54 (2018-11-12): Fix #257

Add missing NUM_DIGITAL_PINS, NUM_ANALOG_INPUTS #defines
commit 2555798c (2018-11-12): Fix #257 

Missing NUM_ANALOG_INPUTS, NUM_DIGITAL_PINS

Also fix the copyright notice to reflect reality.
closed 2018-11-12 by SpenceKonde