SpenceKonde/ATTinyCore #258: attachPCINT not enabled for PCINT8

nseidle, 2018-11-11:
attachPCINT(digitalPinToPCINT(1), updateEncoder, CHANGE);

Works fine but

    attachPCINT(digitalPinToPCINT(8), updateEncoder, CHANGE);

Does not properly map to PCINT8 (pin 10 on dmellis pin numbering). I think I can dig in and fix this but I can't seem to find where attachPCINT is defined? Is it in the Arduino core (couldn't find it)? Or is in in the ATtinyCore? 

For my own notes:

[NicoHood mentions](https://github.com/NicoHood/PinChangeInterrupt): "Pins with * are not broken out/deactivated by default. You may activate them in the setting file (advanced)." PCINT was marked as such - but I cannot find the `setting file`.

This code works:

    PCMSK1 = 1<<0; //Turn on ints for PCINT8
    GIMSK  = bit (PCIE1);  //Datasheet! http://ww1.microchip.com/downloads/en/DeviceDoc/doc8006.pdf

    ISR(PCINT1_vect) { // Interrupt service routine for PCINT8..14 (=ADC0..5)
      updateEncoder();
    }

SpenceKonde, 2018-11-12:
attachPCINT() is not part of the official core, nor mine. So either you are not using my core, or your attachPCINT() function is coming from a library you are using. 
 
I detest functions like attachInterrupt, and attachPCINT is even worse - one of the great advantages of a PCINT is that you run the same ISR for every pin on the that port that is enabled - in some cases this is a great advantage, for example handling rotary encoders - see this: https://github.com/SpenceKonde/UsefulArduinoPosts/blob/master/Encoder_with_PCINT.md ). I recommend doing it the way you did it. 

That said, wherever the attachPCINT function is coming from, if it works on pin1, it should work on pin8 as well. Likely there is a bug in digitalPinToPCINT() - I would be happy to investigate this. Unfortunately, you have neglected to mention which part you are working with, so I don't know where to look. Can you let me know which part you're experiencing this issue with?

SpenceKonde, 2018-11-12:
Assigned to nseidle to provide additional information (which part this problem was experienced with) required for investigation.

SpenceKonde, 2018-11-12:
Update: This core does not define digitalPinToPCINT for any part.

Thus, you are experiencing this problem with a core that is not mine - my core would not work with either version. 

I will consider implementing digitalPinToPCINT macro in a future version of this core.

nseidle, 2018-11-12:
ATtiny84 on Arduino 1.8.7 with ATtinyCore v1.1.5 (looks like I need to update).

nseidle, 2018-11-12:
You're right. Looks like I have NicoHood's library installed from awhile ago and my code was pulling it in. I'll start digging over there. You're welcome to close this issue if you wish. I'll move over there.

I think your interim title change incorrect: Nico's digitalPinToPCINT macro is correct (resolves to 8 correctly), the interrupts are merely [disabled by default](https://github.com/NicoHood/PinChangeInterrupt/blob/master/src/PinChangeInterruptSettings.h#L228);

closed 2018-11-16 by SpenceKonde