SpenceKonde/ATTinyCore #260: ISR (WDT_vect) does not work in ATtiny167 when WDT is enabled

vsv78vrn, 2018-11-13:
ISR (WDT_vect) does not work in ATtiny167 when WDT is enabled. The interrupt bit set. Interrupt allowed. I tried a bunch of code from the Internet. BUT! On ATtimy85 everything works.

vsv78vrn, 2018-11-13:
I use optiboot as a bootloader

SpenceKonde, 2018-11-13:
> wdt_reset(), wdt_enable(wdto_8s) - everything is working! and interruption
> does not occur. What could be the problem?
>
> /* sets the watchdog timer both interrupt and reset mode with an 8 second
> timeout */
> void enableWatchdog()
> {
> cli();
> MCUSR &= (1<<WDRF);
> wdt_reset();
> WDTCSR |= (1<<WDCE) | (1<<WDE);
> WDTCSR = ((1<<WDP1) & ~(1<<WDP2)) | ((1<<WDE) | (1<<WDIE) | (1<<WDP3) |
> (1<<WDP0));
> sei();
> }

Are you using the latest version of the bootloader? This was a known issue
with the virtualboot funtion - on attiny, there is no hardware bootloader
(BOOTRST) functionality, so the bootloader rewrites the vector table and
puts the reset vector in place of one of the other vectors (and the address
of the bootloader in place of reset vector, so bootloader runs on
reset/startup), and jumps to that to start the program. Until ATTinyCore
1.2.0, this used the wdt vector, so the wdt could not be used as an
interrupt source. As of 1.2.0 (released just a few weeks ago), the
bootloader was updated to use the almost totally unused EE_RDY vector
instead.

Connect to ISP programmer and do burn bootloader with the latest version of
ATTinyCore , and it *should* work.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

vsv78vrn, 2018-11-13:
c:\Users\Viking\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.2.1\bootloaders\optiboot\
>>> 1.2.1 <<<

vsv78vrn, 2018-11-13:
Yes, thank you so much, it all worked! Yes, in the place with the update, you also need to sew a new bootloader.

SpenceKonde, 2018-11-16:
Glad it's working!

closed 2018-11-16 by SpenceKonde