SpenceKonde/ATTinyCore #262: Attiny84 with nrf24L01+ (SPI) doesn't work

wavewolf, 2018-11-15:
I'm trying to get my "nrf24L01" module working through SPI with Attiny84 but I can not get it to work.
I am using your core attiny and the library nrf24L01 of this link:
http://tmrh20.github.io/RF24/ATTiny.html

This is my pins:

nRF24L01 VCC, pin2 --- VCC 1|o |14 GND --- nRF24L01 GND, pin1
--...............................--------- PB0 2| |13 AREF
....,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,................ PB1 3| |12 PA1
-----,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,------PB3 4| |11 PA2 --- nRF24L01 CE, pin3
....................................................PB2 5| |10 PA3 --- nRF24L01 CSN, pin4
....................................................PA7 6| |9 PA4 --- nRF24L01 SCK, pin5
nRF24L01 MISO, pin7 --- PA6 7| |8 PA5 --- nRF24L01 MOSI, pin6
+----+

Whay can I do?

SpenceKonde, 2018-11-16:
Can you be more specific about what doesn't work? (I have always had a bad time with the NRF24's)
I am not familiar with that library. Are you using a known good sketch, such as an example provided by the library author? 

Many NRF24 modules are insufficiently decoupled, and need an extra capacitor (~10uF) between Vcc and Gnd, otherwise they lock up the first time you try to transmit. (Also, be aware that the NRF24 is a 3.3v device - it will not work and may be damaged if you apply 5v to it - you probably know this, but lots of people post on the forum and don't seem to, so I figure it is worth mentioning).

wavewolf, 2018-11-16:
I have tried different codes from different libraries of NRF24's and the result has been the same: Or the code gives me error when compiling or loading the code but then it does not work. (it does nothing). I have tried two UNO arduinos with the same NEF24's and the communication works. So, I do not know what to do with attiny84.

SpenceKonde, 2018-11-29:
Did you ever get this sorted out? 

Does it work with a different ATTiny84 core? (try the basic under-featured david mellis core) If that works, but it doesn't work with my core, that indicates a bug in my core that I need to investigate. Otherwise, it indicates a problem somewhere else. 

Thanks

brunoseivam, 2018-12-04:
I *just* got it to work today. RF24 from [here](https://github.com/nRF24/RF24/). Here's my setup:

(note that I had to use the counter-clockwise pin configuration)

``` c++
/*
Core: https://github.com/SpenceKonde/ATTinyCore

ATTiny84, 1MHz, Counter-clockwise pin configuration

                        Ard                     Ard
                        Pin Func  +-\/-+   Func Pin      RF24
 pin2 nRF24L01  VCC ---      VCC 1|o   |14 GND       --- nRF24L01 GND  pin1
                         0   PB0 2|    |13 AREF 10
                         1   PB1 3|    |12 PA1  9
                        RST  PB3 4|    |11 PA2  8    --- nRF24L01 CE   pin3
                         2   PB2 5|    |10 PA3  7    --- nRF24L01 CSN  pin4
                         3   PA7 6|    |9  PA4  6    --- nRF24L01 SCK  pin5
pin 7 nRF24L01 MISO ---  4   PA6 7|    |8  PA5  5    --- nRF24L01 MOSI pin6
                                  +----+
*/

// CE and CSN are configurable, specified values for ATtiny84 as connected above
#define CE_PIN 8
#define CSN_PIN 7

#define LED_PIN 0

#include <RF24.h>

const byte ADDR[6] = "00001";

RF24 radio(CE_PIN, CSN_PIN);

void setup() {
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  // Blink fast twice before sending
  digitalWrite(LED_PIN, HIGH);
  delay(100);
  digitalWrite(LED_PIN, LOW);
  delay(100);
  digitalWrite(LED_PIN, HIGH);
  delay(100);
  digitalWrite(LED_PIN, LOW);

  char buffer[32] = {};
  short *battery_raw_ptr = (short*)&buffer[0];
  
  // Read internal reference against VCC reference (default)
  // 1.1V internal reference = 0b100001 = 0x21
  short battery_raw = analogRead(0x21); // 1.1V internal reference = 0b100001
  // The battery level, in volts, will be: 1024*1.1*1000 / battery_raw;

  *battery_raw_ptr = battery_raw;
      
  radio.begin();
  radio.openWritingPipe(ADDR);
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_250KBPS);
  radio.stopListening();
  radio.write(&buffer, 2);

  radio.powerDown();
  
  delay(3000);
}
```

SpenceKonde, 2018-12-04:
Thanks for the update.

closed 2018-12-04 by SpenceKonde