SpenceKonde/ATTinyCore #263: 128KHz breaks delay()?

WereCatf, 2018-11-24:
I'm not entirely certain if I'm just doing something wrong or if I've misunderstood something, but my Attiny85 seems to crash/stall if I call delay() when using internal 128KHz clock. Absolutely nothing happens after I've issued delay(1) or any other value.

sleemanj, 2018-11-24:
The minimum clock speed is probably 1MHz if you want
millis/micros/delay/delayMicroseconds to be sensible... 

https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/Arduino.h#L66

https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/wiring.c#L218

https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/wiring.c#L410

With that said, I would have thought it would work, just take about 10
times longer than expected (delay(1) would be more like delay(10))

sleemanj, 2018-11-24:
[delay() uses micros() to
calculate](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/wiring.c#L276)

Using millis() and a spin wait might be accurate enough for you, in
which case you could simply create your own delay function 

    void myDelay(uint32_t howLong)
    { 

      uint32_t start = millis();
      while((millis() - start) < howLong)  /* NOP */;
      return; 

    } 

On 2018-11-25 09:29, WereCatf wrote:

> millis() does work, though! Haven't tested micros(), but delay() is broken.

WereCatf, 2018-11-24:
@sleemanj Yeah, I am already using a millis()-based workaround. I just decided to raise an issue with other people in mind; even if this never got fixed, at least it's documented now.

SpenceKonde, 2018-11-24:
Which part is this? There are two cores included in this package depending
on which attiny is in use.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

WereCatf, 2018-11-24:
@SpenceKonde Ah, I forgot to mention: I am using an ATTiny85.

SpenceKonde, 2018-11-29:
Why is micros() giving wrong answers? (that's the only way that delay() should fail here)
Shouldn't this work? 

`#if F_CPU < 1000000L
return ((m << 8) + t) * MillisTimer_Prescale_Value * (1000000L/F_CPU);
#else`

SpenceKonde, 2018-11-29:
Nevermind, I did it. 

But micros() looks close to correct......

SpenceKonde, 2018-11-29:
I'm baffled, delay *ought* to be working!

SpenceKonde, 2018-11-29:
AAHA! 
If it jumps more than 1 ms on the last run through the inner loop, it would roll over ms! 
Added a failsafe against that, since it could happen in other cases under extenuating circumstances. At least now delay won't be a permahang if that ever happens. 
For sub-1MHz clock speeds, delay() will now use millis() based implementation.

commit 13f5e953 (2018-11-29): Fix #263
closed 2018-11-29 by SpenceKonde