SpenceKonde/ATTinyCore #273: Attinyx41 - i2c slave and sleep

battosai30, 2018-12-05:
Hi,

I built a PCB with Attiny841 to get a Maxbotix MB7683 ultrasonic sensor very low power and i2c connected. The Maxbotix power is driven by a Mosfet and Attiny reads serial message and allow an i2c master to interrogate it.

The schema is : 
- Master send a command to launch a new measure.
- The slave receive with command, manage all it has to do to initiate a new measure. During this phase, as a complete new measure needs ~500 ms, a status flag is available to check if the new measure is available or not.
- The master polls the slave to get the status flag (with a resquest). As soon as the slave set the flag to 0, the master requests the last measure.
- Out of this phases, Attiny is sleeping.

So my issue : sleep as it is OK, and wake up interrupt from i2c adress matching seems ok too. My issue is there is no status flag to know if i2c communication is acting or not, and if not, go to sleep in my case. More precisely : as there is no Wire.flush() when Attiny is starting to respond to a request, the receive callback is called, puts the datas in buffer but as it's a non blocking function, when it comes back to the loop() with my sleep function, the Attiny go to sleep without finishing his transmission.

I simplified my original code to get one that exactly leads to the same issue : 

```
#include <Wire.h>
#include <avr/sleep.h>
#include <avr/interrupt.h>

void setup() {
  
  Wire.begin(8);
  Wire.onRequest(requestEvent);

}

void loop() {
  
  goToSleep();

}

void requestEvent()
{
  byte buf[] = {1, 2};
  Wire.write(buf, 2);
}

void goToSleep()
{

  (ADCSRA &= (~(1 << ADEN)));
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  interrupts();
  sleep_cpu();
  sleep_disable();

}
```
If I simply comment the goToSleep() call, everything is OK. If not, the master is stuck (I did not check with a logic analyzer but I guess the data line is held LOW).

I tried to implement and work with onStop() event but does nothing.

Thanks

SpenceKonde, 2018-12-05:
A stopgap measure would be to set a flag in the ISR, and then if the flag is set, wait 500ms (or some known safe length of time) for the transmit to finish. 

Wire.flush() is unimplemented for ATTiny841 and ATTiny828 - it looks non-trivial; I don't think I can get it into the 1.2.2 update without unduly delaying it.

battosai30, 2018-12-09:
Hi, I tried to implement a flag but it doesn't seems to work (or I made a bad code).

I found a workaround for now : I simply implemented an i2c sleep command, so the slave doesn't go sleeping by his own, the master do it.

SpenceKonde, 2018-12-11:
Updated issue to reflect planned changes.

SpenceKonde, 2018-12-12:
Hm. This doesn't seem to be implemented anywhere - the official Wire library doesn't even implement it.

SpenceKonde, 2020-04-08:
Closing with no plans to change. As there is no standard behavior here, and I have many other fish to fry, it doesn't make sense to leave this in the queue. At some point (probably when I do a deep dive into I2C on the megaavr parts) if I decide that Wire.flush() is necessary, I'll figure out what the right behavior for it is, implement it, and then port it to this core. But if/until that happens, no plans.

closed 2020-04-08 by SpenceKonde