SpenceKonde/ATTinyCore #274: Is it possible to have an input change interrupt on Tiny48 port PA2?

zapta, 2018-12-14:


zapta, 2018-12-14:
I am trying to setup a on change interrupt on PA2 (using AtTiny 32 pin package) and am not sure if this is supported here. I don't mind doing some registry manipulation that bypasses the Arduino API.

Can you please provide a few hints?  

I looked at the software serial interrupts implementation but am not sure if it will work for PA2 since the core seems to define only two (?) interrupt vector and PA2 is associated with PCINT26.

SpenceKonde, 2018-12-14:
There's nothing special about setting up pin change interrupts - the datasheet contains exhaustive information on how they work; see chapter 9 "External Interrupts"

Do not confuse INTx interrupts (INT1, INT0) with pin change interrupts. The abomination that is arduino attachInterrupt() is concerned with the INTx interrupts - it has nothing to do with pin change interrupts. Pin change interrupts are in banks of 8 - one ISR for each port, and they only can fire on change (rather than rising/falling/low level for the INTx interrupts), but they are handled async so you can use them to wake from sleep (INTx can only wake from sleep on low level). With PCINTs the responsibility is on you to, in the ISR, figure out which of the 8 pins changed (if more than one are enabled for interrupt in the corresponding PCMSKn register) and what it changed to/from, which is nothing complicated (though I would always do direct port reads here, for speed) - in some cases this actually makes the overall task much easier (for example, handling multiple rotary encoders). 

To use them define the ISR for PCINTn_vect (where n is 0~3), and then to enable it, you need to set the bits in the PCMSKn corresponding to the pin(s) within the bank that you want to enable it for, and the the PCIEn bit in PCICR (you can disable it by unsetting either of these). IE, for PA2
ISR(PCINT3_vect) {
//do your ISR here
}

and where you want to enable it:
PCMSK3=1<<PCINT26; //(I think this will work - PCINT26 should be #defined to 2. But if it isn't, PCMSK3=1<<2; will work). 
PCICR |= 1<<PCIE3; 

As you can see in https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/libraries/SoftwareSerial/SoftwareSerial.cpp we do define ISRs for all the PCINT vectors, so the software serial library works on  all pins.
 
Note that if you want to use serial, there is a built-in software serial implementation called Serial (for compatibility with code written for chips with a hardware serial port) See section Serial under https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x8.md - because this makes use of the ACO vector instead of PCINTs, you can use it at the same time as you're using the PCINTs for something else.

SpenceKonde, 2018-12-14:
Closing issue as does not represent an issue with the core, but feel free to continue to reply here.

closed 2018-12-14 by SpenceKonde

zapta, 2018-12-14:
Thanks Spence. Very good explanation and everything works now. I needed to use PA2 input transitions to wake up a Tiny48 from sleep mode and bypassing Android and working directly with the registers per your instructions did the trick. 

BTW, your Tiny48 board arrived today. Thanks!

SpenceKonde, 2018-12-15:
Yay, glad to help. And thanks for your order :-)

zapta, 2018-12-17:
Your Tiny48 Core is now running the soap dispenser in our kitchen.  ;-)

https://github.com/zapta/misc/tree/master/soap_dispenser