SpenceKonde/ATTinyCore #275: ADC Reference

smartroad, 2018-12-21:
Hi, not an issue, but I have been looking for an answer but not found one that makes sense to me ;) How do I set the ADC reference to 1.1V? I can't figure out how.

SpenceKonde, 2018-12-21:
Just do:
analogReference(INTERNAL); 

(in place of INTERNAL, you can use any of the options listed in the part-specific documentation page for the part you're using - not all boards have the same options)

SpenceKonde, 2018-12-21:
Closing because not an issue with the core, feel free to comment if you have additional questions about this.

closed 2018-12-21 by SpenceKonde

cms-codes, 2018-12-21:
It varies depending on the particular MCU, but usually involves setting
certain bits in the ADC register.

I'd suggest checking out the datasheet for the target hardware you're
using. Microchip/Atmel's documentation is well laid out and you should be
able to find details on the internal reference.

SpenceKonde, 2018-12-21:
@cms-codes - that won't work if using the analogRead() to read it; use analogReference() like I said above. This is the same as the official arduino core. Setting the reference via the registers will get set back to what was last set with analogReference() (or the default reference, Vcc if it hasn't been set already) the moment you call analogRead().

cms-codes, 2018-12-21:
Right. I am used to going about this directly rather than using the
analogRead() function. Apologies for the confusion.