SpenceKonde/ATTinyCore #276: usb programmer pins

DaveCalaway, 2018-12-23:
I everyone,
i recently bought a usb programmer https://goo.gl/ixTHbH .
I checked the connection, https://goo.gl/49LWK2 , and i immediately realized that this usb programmer uses the PB4 and PB3 to program the board ( micronucleus bootloader ).

I read the references for the ATTinyCore bootloader for the tiny85, it uses the pins PB0 and PB1.

Can i burn the micronucleus bootloader and flash the code without the ATTinyCore bootloader?
Or it doesn't works?

SpenceKonde, 2018-12-24:
This core only supports the included optiboot serial bootloader; there is
an open issue (with no planned version) for adding support for the
micronucleus/digispark bootloader, but there is not presently support for
that, and it is considered a low development priority (i have a low opinion
of vusb bootloaders - they are space hogs, and have compatibility issues
with some usb hosts, because they arent quite compliant with the usb
standard, and require external components that nobody tells you about the
ramifications of, ex. on digispark boards that integrate the parts on that
"programmer" and chip onto one board, with no way to remove the chip). I
often just use isp programmers and no bootloader to maximize available
flash.

Also, despite what that vendor calls it, that is not a programmer - thats
just the external components needed for vusb and a usb socket; you need a
chip with a vusb bootloader on it already. A programmer would have another
chip on itt that could program the target chip (for avrs, via ISP - for
example a usbasp or usbtinyisp - both are dirt cheap and work great), and
could be used to put the bootloader on a virgin chip and set the fuses.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2018-12-29:
Closing as this is not an issue with the core.  See issue #210 regarding digispark/micronucleus support.

closed 2018-12-29 by SpenceKonde