SpenceKonde/ATTinyCore #277: impossible to compile

Corruped, 2018-12-30:
Hi , every time that I tried to compile a sketch , and there is a strange compilation error (even with arduino as ISP (ATTinyCore) sellected).
here the error message  https://pastebin.com/a4bPhi7T

closed 2018-12-30 by Corruped

SpenceKonde, 2018-12-30:
Ah, yeah, the windows store version has a wide variety of mysterious defects. Always use the real version. 

You should report this issue to the Arduino IDE team, as it indicates a defect in the Windows Store version.

per1234, 2018-12-31:
It might be this: https://github.com/arduino/Arduino/issues/7949. That bug actually affects the standard Arduino IDE on Windows as well as the Windows Store version.

This bug is specific to the 5.4.0-atmel3.6.1-arduino2 version of avr-gcc used by Arduino AVR Boards 1.6.22 and newer. It occurs sporadically for me, coming and going with no rhyme or reason.

The workaround is to roll back to Arduino AVR Boards 1.6.21. There is also a beta version of the new AVR Boards package, which uses avr-gcc 7.3.0 and is supposed to solve that bug:
https://github.com/arduino/Arduino/issues/7949#issuecomment-442969335

pkoevesdi, 2019-01-28:
I can confirm that behaviour: No Compiling within Arduino IDE Windows Store Version 1.8.8 and ATTinyCore 1.2.2. Error message with me says only "access denied".
Works fine with Arduino IDE 1.8.8 installed from .exe.
Don't know why this issue is closed and the hint in the readme has gone. I think, it still belongs there.

SpenceKonde, 2019-01-30:
Okay, will reinstate the warning.