SpenceKonde/ATTinyCore #279: SoftwareSerial BUG

Corruped, 2019-01-02:
Hi I'm currently on **desktop** Arduino IDE
and **1.2.2 AttinyCORE**

when I try to compile my sketch with Some _Softwareserial_ stuff in it I get this error https://pastebin.com/BQxmQSuh

Corruped, 2019-01-02:
here's my code btw https://pastebin.com/4bE94DVR

Corruped, 2019-01-02:
yeah i know but I dint thought about it when i designed my PCB  so yeah I'll change that (i can still fix my prototype)

SpenceKonde, 2019-01-07:
Fixed - there was a typo in the macro used when the new-style pinout is used. 

As an aside, your code still doesn't compile because you're using PCINT and software serial at the same time. Software serial uses PCINT vectors for handling incoming data, so you can't use PCINTs in your own application - one of many reasons to avoid software serial.

closed 2019-01-07 by SpenceKonde