SpenceKonde/ATTinyCore #280: ATtiny2313A error: 'DDA0' was not declared in this scope

helgevh, 2019-01-06:
Hi, 
I am using the latest version of ATTinyCore and Arduino IDE 1.8.8. While addressing the ATtiny2313A pins with the Arduino pinMode just works fine, addressing the registers DDRB, DDRD and DDRA only works for the first to. The command for DDRA results in an error message.

`// LED Pins
 
void setup() {
  // Pin configuration input/output
  DDRB = (1<<DDB1)|(1<<DDB0)|(1<<DDB2)|(1<<DDB3)|(1<<DDB4);
  DDRD = (1<<DDD1)|(0<<DDD2)|(1<<DDD3)|(1<<DDD4)|(1<<DDD5)|(1<<DDD5);
  DDRA = (1<<DDA0);
 
}

void loop() {
  // put your main code here, to run repeatedly:

  PORTB = (1<<PB1)|(1<<PB0)|(1<<PB2)|(1<<PB3)|(1<<PB4);
  PORTA = (1<<PA0);
  PORTD = (1<<PD1)|(1<<PD3)|(1<<PD4)|(1<<PD5)|(1<<PD5);
}`

The error message looks like this:

`C:\Users\Helge\Documents\Arduino\sketches\Pollin-C-Code\Pollin-C-Code\Pollin-C-Code.ino: In function 'void setup()':

Pollin-C-Code:7:14: error: 'DDA0' was not declared in this scope

   DDRA = (1<<DDA0);

              ^

exit status 1
'DDA0' was not declared in this scope
`
The same error is shown for DDA1 and DDA2.

SpenceKonde, 2019-01-07:
This is a bug in the atmel-supplied io2313.h file!

`/* Port A Data Direction Register DDRA[2:0] */
#define DDRA    _SFR_IO8(0x1A)

#define DDRA2   2
#define DDRA1   1
#define DDRA0   0
`

SpenceKonde, 2019-01-07:
Fixed in github version now (I just provide the missing defines if they're not there now). Code now compiles.

closed 2019-01-07 by SpenceKonde

helgevh, 2019-01-07:
I just took a look in the io2313.h file. Looks like its made for the older AT90S2313 which doesn't have a Port A.