SpenceKonde/ATTinyCore #281: SoftwareSerial power consumption

DaveCalaway, 2019-01-08:
Hi everyone,
i written a code that wake-up, from the deepSleep, the tiny85.
The pin interrupt is set by "change pin interrupt".
After it's wake-up, sends a number over the Serial with SoftwareSerial ( TX is AIN0, RX is AIN1 ).
I disabled the BOD and it's run over 1Mhz under 3.3V.

Now if i connect the RX_tiny to TX_ardu and TX_tiny to RX_ardu, the multimeter increases the current consumption, up to 3.6 mA from less than 20uA. 
The tiny stay in the deepSleep mode, because if i detach the RX and TX connection the consumption falls on 20uA.

Is it an hardware problem, based on the Serial circuit on Arduino and Tiny85 or i miss something?

DaveCalaway, 2019-01-09:
> Is the arduino it's connected to a 5v one without a level shifter? I would expect to see significant idle current through the protection diodes in that case - this can also damage the chip. If doing this, you need a level shifter (they're under a buck shipped from ebay).
Oh my mistake, you are right.
Using the level shifter, all works fine.

If you don't have the converter, you can use the 5v instead the 3v3 ( it's consume a little more, but for the test is ok ).

> That's my first guess. If not...
> Are you using the builtin software serial implementation (the one named Serial) or the SoftwareSerial library with those same pins? Does it happen if you do the other one?

Yes i think so.
I'm using the ATTinyCore on the PlatformIO, using this guide: https://goo.gl/H4gJRJ
Valeros - https://goo.gl/86UTRX - added the ATtinyCore on the PlatformIO.
Now, who updates the core in PlatformIO compared to the official one?
I checked the ATtinyCore on the PlatforIO, it has been updated 3 months ago.

SpenceKonde, 2019-01-10:
It should be entirely possible to get the power usage lower than that while in sleep mode.....

I wonder if the power consumption when using the buitlin software serial named Serial is worse because it enables the analog comparator? 

I do not know who is responsible for updating ATTinyCore in PlatformIO - i think last time it was done, it was done in response to nagging from some user. There have been big updates in ATTinyCore since then though (including serial bootloaders for all parts with an 8k version (including the 4k versions of those chips, but not chips that only come in 2k/4k versions), support for additional parts, and a ton of important fixes.

SpenceKonde, 2019-01-10:
Do you get better power consumption using SoftwareSerial library compared to the builtin software serial? If so, I will be sure to document that limitation. 

Closing issue as there is no bug in the core here - but do let me know if SoftwareSerial library results in lower power consumption. If it does, it just needs to be documented (this would make it a tradeoff of sleeping power consumption vs use of PCINTs)

closed 2019-01-10 by SpenceKonde

SpenceKonde, 2019-01-10:
If the platformIO people say that there's anything I need to do to get ATTinyCore for PlatformIO updated let me know.

DaveCalaway, 2019-01-10:
> Do you get better power consumption using SoftwareSerial library compared to the builtin software serial? If so, I will be sure to document that limitation.

Sure! But i can try it only in the 8Mhz mode. Under 8Mhz is not works good: https://goo.gl/nJHtov

> If the platformIO people say that there's anything I need to do to get ATTinyCore for PlatformIO updated let me know.

Looks that your ATtinyCore, for these boards https://goo.gl/7U1cq4, is the main core for PlatformIO.
The lables "tiny" https://goo.gl/3xi1M5 is linked here https://goo.gl/G2jqZL under the root https://github.com/platformio/platformio-pkg-framework-arduinoavr .
Probably someone update your core under PlatformIO...

DaveCalaway, 2019-01-11:
Very strange results: https://goo.gl/wzdnt2 

SoftwareSerial |  Serial on    |     Sleep
|  ------------ | ------------- | -------------
ATtinyCore|  885uA       |     21,21uA
Arduino  |  865uA       |     4,65uA

The Arduino's SoftwareSerial is in line with the power consumption, yours is not... but why? 
The example is identical... 
[image]