SpenceKonde/ATTinyCore #282: tinyX61 SPI pins are mixed up

targodan, 2019-01-10:
In the [tinyX61 variant file](https://github.com/SpenceKonde/ATTinyCore/blob/53b203fb8388acd7dc80e597b9f361b1127fb4f5/avr/variants/tinyX61/pins_arduino.h#L53) the DI and DO pins are mixed up. According to the [datasheet](http://ww1.microchip.com/downloads/en/devicedoc/atmel-2588-8-bit-avr-microcontrollers-tinyavr-attiny261-attiny461-attiny861_datasheet.pdf) PB0 is DI and PB1 is DO. You've got it mixed up.

**Wrong:**
```cpp
#define DO_DD_PIN DDB0
#define DI_DD_PIN DDB1
```

**Right:**
```cpp
#define DO_DD_PIN DDB1
#define DI_DD_PIN DDB0
```

This leads to nasty failure of any SPI communication. I tested it on my attiny861a and have had successful communication with my version (right) whereas the communication was unsuccessful without the fix.

targodan, 2019-01-11:
I just opened a pull request fixing this.

SpenceKonde, 2019-01-12:
Much thanks for not only reporting this, but providing a PR!

closed 2019-01-12 by SpenceKonde