SpenceKonde/ATTinyCore #284: Newer version of ATTinyCore not working

gcormier, 2019-02-03:
I have a project that when using the newer version, no longer works.

I've tried both the clockwise and counter-clockwise options with no luck. Is there anything else I should be checking? Using 1.1.5, it works perfect.

SpenceKonde, 2019-02-03:
What are the symptoms? There may have been a regression that I haven't fixed yet (because I'm not aware of it), but without more information, I couldn't venture a guess

Do you get a compile error? If so post the compile output.

SpenceKonde, 2019-02-03:
But the behavior doesn't match? 

That's... not good at all. Can you post the code or a link to it, and a description of what it used to do, and how that differs from what it does now?

SpenceKonde, 2019-02-03:
My guess is that whatever broke, it's 1.2.2 that did it, not 1.2.1. (1.2.0 and 1.2.1 are the same for everything except the tiny x61)

gcormier, 2019-02-03:
This is the project

https://github.com/gcormier/megadesk

Unfortunately without that desk it would be hard to test, but you can see the code if it gives any hints.

The part that "broke" was the LIN bus communications. This is handled in lin.cpp.

gcormier, 2019-02-03:
1.2.1 doesn't work. Trying 1.2.0...

gcormier, 2019-02-03:
Okay, 1.2.1 does work. 1.2.2 does not.

SpenceKonde, 2019-02-03:
Can you try with the github version? I just fixed a really nasty mistake relating to the 841

Ie, uninstall via board manager, then sync this repo into (sketchbook folder)/hardware (or download it, extract it and put the folder there, renaming it to ATTinyCore)

SpenceKonde, 2019-02-03:
Especially if that fixes it, I will do a new release tonight. That was a horrifying bug that I'm astonished wasn't caught by *someone* before now.

gcormier, 2019-02-03:
So I should have

megadesk\megadesk.ino (and other source)
megadesk\hardware\ATTinyCore\avrdude_conf.txt etc

Then how do I select it in the Arduino IDE?

SpenceKonde, 2019-02-04:
No, that's not the correct location. 

Your "Sketchbook folder" is the folder listed at the top of File -> Preferences (typically C:/Users/YourUsername/Arduino  - but can be set to other locations)

SpenceKonde, 2019-02-04:
Er, sorry, C:/Users/YourUsername/Documents/Arduino is the default

SpenceKonde, 2019-02-04:
Should be Documents/Arduino/hardware/ATTinyCore (you will have to create the hardware folder if you have not previously manually installed any board packages)

gcormier, 2019-02-04:
Yes, your fix worked!

SpenceKonde, 2019-02-04:
Yay! The commit that fixed it was 77e367a51255e8a76d5278eab079dc266f34cbd7

closed 2019-02-04 by SpenceKonde

gcormier, 2019-02-04:
How long does it normally take to show up on board manager?

SpenceKonde, 2019-02-04:
It's in board manager now