SpenceKonde/ATTinyCore #285: Add migration documentation

SpenceKonde, 2019-02-08:
It has been brought to my attention that documentation on migrating to ATTinyCore from other arduino cores, and to attiny841 from attiny84 is needed. 

* Migration to ATTinyCore from other ATTiny hardware packages (damellis' and other) - main differences being the builtin software serial implementation and universal wire/spi libraries. 

* Migration to '841 from '84 - expanded timer functionality, SPI/I2C vs USI, hardware serial vs software serial, differences in oscillator temperature/voltage dependence. 

* Migration from an ATmega (ex, uno/nano/etc) to ATtiny device, general and specific concerns.

becky001, 2019-02-08:
I made a side by side spreadsheet comparing all the registers of Attiny84 and Attiny841.  Maybe someone will find this useful.
[Attiny84_841_Reg_Compare.xlsx](https://github.com/SpenceKonde/ATTinyCore/files/2845498/Attiny84_841_Reg_Compare.xlsx)

SpenceKonde, 2019-02-12:
I've added some notes here. Let me know or give pullrequest if there's anything missing.

closed 2019-03-10 by SpenceKonde