SpenceKonde/ATTinyCore #286: Multiple I2C slaves

vitpetrik, 2019-02-10:
Hi,

I need to get working circuit with multiple ATtiny85 slaves. The master is ESP32. I'm having no issues with running only one slave, but with more slaves the communication fails.

the ESP outputs this through serial port

`[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=1`

slave code running on ATtiny85 at 8MHz
`#include <Wire.h>

int i = 0;

void setup() {
    Wire.begin(7);
    Wire.onRequest(requestISR);
}

void loop() {}

void requestISR() {
        Wire.write(i);
        i++;
        if(i > 250){
          i = 0;
        }
}`

Thanks for all our help and amazing job with the arduino core

vitpetrik, 2019-02-10:
Sorry... I'm total stupid. I'm learning with freeRTOS and made an error on the ESP32 side.

Again, sorry for interrupting you and thanks for the amazing job you've done

SpenceKonde, 2019-02-11:
Glad you got that sorted! 

Particularly since I really didn't want to set up a multiple I2C slave testbed.

closed 2019-02-11 by SpenceKonde