SpenceKonde/ATTinyCore #287: support for attiny167 broken?

tomaskovacik, 2019-02-12:
hello,
as I was trying to investigate if digispak pro will work with attinycore I realize that support for this mcu is probably broken or I'm doing something wrong ...

I try to simple blink sketch using delay from examples with pin name PB1 (https://s3.amazonaws.com/digistump-resources/files/d312a4a8_ProFinal.pdf)

void setup() {
  pinMode(PB1, OUTPUT);
}
void loop() {
  digitalWrite(PB1, HIGH); 
  delay(1000);
  digitalWrite(PB1, LOW);
  delay(1000);
}

while using digistump support in arduino and upload using SPI or over micronucleus usb bootloader everything works just like it should.

Same sketch compiled with attinycore and uploaded over USBasp did not work :/

I have selected : 
 "16Mhz (extenal)" oscilator in board menu
 mcu ATtiny167
BOD and LTO disabled 
eeprom I try both
ping maping: I try both

logs if it help with anything:

[Digistump.txt](https://github.com/SpenceKonde/ATTinyCore/files/2855658/Digistump.txt)

[ATTinyCore.txt](https://github.com/SpenceKonde/ATTinyCore/files/2855659/ATTinyCore.txt)

i'm doing something wrong?

SpenceKonde, 2019-02-12:
You cannot use PB1 or the other Pxn pin names with digitalWrite or any
other similar functions. Not on attinycore or any other arduino core. They
are defined by the files provided by atmel for use with direct port writes,
each defined as the number of the pin within the port. For digitalWrite et.
al. you must use the arduino pin number shown in the pinout chart, or the
PIN_B1 type constants provided by the core (which are defined to the
arduino pin number of that pin). On the attiny85, the Pxn defines happen to
work because there is only one port, and the numbers of the pin within that
port match the arduino pin numbers. It doesnt work with any other part, nor
could it work on any part with more than one port.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

tomaskovacik, 2019-02-12:
hello, 

thx for clarification, strange it works sometimes(but probably only in stm32 environment), but this was  for sure reason of long nights full of debuging without progress:)

ths again.

t.

ps: links to pin mapping pictures for tiny167 and tiny87  in README.md are pointing to non existing files

closed 2019-02-12 by tomaskovacik

tomaskovacik, 2019-02-12:
ok, works, also problem was mapping, in my digispark PRO support branch I use tinyX7 but tinyX7_New should by used

thx again