SpenceKonde/ATTinyCore #295: Issues with Programming the Attiny841

witnessmenow, 2019-03-02:
Hey,

First, thanks so much for all the work on this library!

I was playing around with programming an 841 last night and I ran into some issues that I was wondering if you could help me figure them out.

## My Setup

- I'm using Arudino IDE 1.85 on Windows 10 (I'm using a [portable version](https://www.arduino.cc/en/Guide/PortableIDE) if that helps)
- I upgraded Arduino AVR boards to 1.6.21 (as you mentioned in the readme about problems with 22 and 23)
- I installed this from the [boards manager](https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md#boards-manager-installation) method
- Using an Arduino Nano as ISP (flashed with Arduino ISP sketch)

## What worked for me:
I did eventually get this working, so lets start with what did work for me!

- Selecting "841 (Optiboot)" for the board, Selecting "Arduino as ISP" (note: not the one with "attinyCore" in the name) and burning the bootloader
- I could then use a regular USB to serial adapter to program via TX0 and RX0 pins of the 841

## What didn't work for me

### "Burn bootloader" option when "Arduino as ISP (AttinyCore)" was selected

This didn't with either 841 board type (no bootloader or optiboot), verbose logs below 

Is this expected? "Arduino as ISP" worked fine for "Burn bootloader" but all references I seen in the documentation mentions I should use the (AttinyCore) one

```
System wide configuration file is "C:\Users\Brian\Documents\ArduinoIDE\1.85 with RC2.5\arduino-1.8.5\portable\packages\ATTinyCore\hardware\avr\1.2.3/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 5.0
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15
         AVR Part                      : ATtiny841
         Chip Erase delay              : 15000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     6     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash         65    12    32    0 yes      8192   16    512  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 4728160
         Firmware Version: 0.4598883
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

Error while burning bootloader.
```

### Programming via ISP when "(No Bootloader)" board was burned

As mentioned above, using "Arduino as ISP" I was able to burn the "Attiny441/841 (No Bootloader)" bootloader (confusing sentance :))

But I was unable to program the board when this board type was selected.

With "Arduino as ISP" selected I was getting the 'AVR part "attiny841" not found' message (which is expected I guess)

With "Arduino as ISP (AttinyCore)" selected, I get a "the selected serial port does not exist or your board is not connected" error, despite it being the same serial port I used to burn the bootloader. Full output below

```
System wide configuration file is "C:\Users\Brian\Documents\ArduinoIDE\1.85 with RC2.5\arduino-1.8.5\portable\packages\ATTinyCore\hardware\avr\1.2.3/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 5.0
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15
         AVR Part                      : ATtiny841
         Chip Erase delay              : 15000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     6     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash         65    12    32    0 yes      8192   16    512  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

the selected serial port 
 does not exist or your board is not connected
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 4728160
         Firmware Version: 0.4598883
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

```

If you need any more info please let me know.

Brian

SpenceKonde, 2019-03-07:
FFS.....

It should ALWAYS give the error that attiny841 was not found if the (attinycore) version is not selected when doing Burn Bootloader, and probably Upload using Programmer. It should never work. The fact that it sometimes does indicates strange inconsistent behavior of the IDE w/regards to whether it respects platform.txt when a programmer supplied by the IDE is selected. My understanding is that (despite any logic to the contrary) it should ignore the platform.txt of the board package corresponding to the selected board if the programmer was supplied by the default hardware package. So the fact that the non-ATTinyCore version of the programmer ever works makes no sense. 

It should require the 10uF capacitor between Reset and Ground on the Arduino being used as ISP, otherwise it will get the resp 0x14 0x02 0x10 error. This happens because the protocol is set to arduino, which I was told I had to do to make it compatible with something else, but when that's set, it will autoreset the board if the cap isn't in place. 

I cannot reconcile my understanding of how the IDE is supposed to work with what you are seeing. 

Anyone know if there is a way to make it so the IDE only shows programmers supplied by the hardware package corresponding to the selected board, and doesn't show programmers supplied by the hardware package when other boards are selected? This would make life simpler.

Floessie, 2019-03-07:
USBAsp: I can recommend [this firmware](https://github.com/bperrybap/usbasp/tree/1.06-alpha/bin/firmware/usbasp-v1.06-alpha-2016-05-18) for the cheap clones. Get at least two of them and update them mutually.

SpenceKonde, 2019-03-10:
I'm pulling in the latest approach that the official core used.

closed 2019-03-11 by SpenceKonde