SpenceKonde/ATTinyCore #308: Question: UART Baud Rate Precission

tatobari, 2019-04-06:
Does the following quote mean, for example, that if I want to work with 16mhz at 115200bps with an external crystal I need to choose 14.7456 MHz because 16mhz won't be precise and I may have sync issues?

> External crystal (all except 828, 43 and x8 series):
> 
> 20 MHz
> 18.432 MHz*
> 16 MHz
> 14.7456 MHz*
> 12 MHz
> 11.0592 MHz*
> 9.216 MHz*
> 8 MHz
> 7.3728 MHz*
> 6 MHz
> 4 MHz
> All available clock options for the selected processor will be shown in the Tools -> Clock menu. Options marked with a * are "UART frequencies", used to get perfect match to common UART baud rates. These are not available for chips without a UART in versions of ATTinyCore older than 1.1.5.

Thanks!

sleemanj, 2019-04-06:
According to my build scripts, 16MHz will result in a -2.1% error.  If you are using an external crystal, this is probably OK, if you were using an internal oscillator (PLL whatever) it might be a bit iffy.

SpenceKonde, 2019-04-06:
16MHz is fine for 115200. That's what the Arduino Uno/Nano/ProMini do. 

The "UART frequencies" are for applications where people insist on perfect match to a certain baud rate. But for typical applications you don't need it that close.

closed 2019-04-06 by SpenceKonde

tatobari, 2019-04-08:
One more questions, how are the values of `UBRR` and `U2Xn` set?

SpenceKonde, 2019-04-08:
U2X is used* - see https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tinymodern/HardwareSerial.cpp for the actual code used to calculate UBRR

* Because the hardware serial code is a verbatim copy of the official one, it has the same odd hardcoded exception to U2X (16MHz and 57600 baud doesn't get U2X) that the official one does. I'm creating an issue about that; I don't think we need to do that here.

tatobari, 2019-04-09:
Thanks man! As far as I can see, the code resolves to the UBRR with the smallest error for the intended baud rate. During the weekend I realised I was using an 8Mhz crystal and not a 16Mhz one. Communication error with the guy who's designing the circuits. I'm gonna change the crystal and try again at 115200 and 16Mhz. Thanks a lot for the help!

commit 80d6bc41 (2019-04-06): Improve UART frequency documentation. (fix #308)

Issue #308 made clear that the documentation could lead people to incorrect conclusion about these.