SpenceKonde/ATTinyCore #310: Attiny84 with 128 kHz oscillator

ideaalab, 2019-04-12:
Is it possible to set the fuses so the Attiny84 can run with the 128 kHz internal oscillator?
For low power/low speed applications this can save some uA.
Thanks!

sleemanj, 2019-04-12:
In my fork ( https://github.com/sleemanj/optiboot/blob/master/dists/README.md#attiny ) you can choose the 128kHz oscillator from the "Tools > Override Clock Source" menu and then burn bootloader to set the fuses for it.

**However** generally speaking below 1MHz or below things may not work and I would not recommend it, even if the rest of the code was OK with running that slow the 128kHz oscillator is not likely to be anything close to accurate.

Better in my opinion to run at a reasonable speed and just go to sleep as deep and as much as possible - my [SimpleSleep Library](https://github.com/sleemanj/SimpleSleep) makes sleeping, err, simple.

ideaalab, 2019-04-12:
Thanks!  I will try your fork :)
Do you know if @SpenceKonde is willing to merge your oscillator configurations into AttinyCore?
My code just turn on and off some LEDs according to some switches. Accuracy of timing is not really needed. Your sleep library looks very nice!

ideaalab, 2019-04-12:
By the way, ive just checked the ATtiny84 datasheet, and 8Mhz consume more than 10 times more current than 1Mhz. And 1Mhz is about 5 times more than 128khz (quick glance at the active supply current on the datasheet)

SpenceKonde, 2019-04-12:
Thanks for reporting this issue. 128kHz was supposed to be supported for an option with all ATtiny devices that support it (for non-optiboot configurations). Some chips only have a 32kHz ULP. Apparently when I added support for it though, I only added it to the x5 series. It is now available on all parts. I have just added support for it (you'll have to uninstall the board manager version, and install the github version to use it before the next release - I use git to sync the repo to a folder in hardware)

freemovers, 2019-04-12:
For switching LED's based on some switches I would use the interrupt pin (int0) and keep the ATtiny in sleep mode the rest of the time. If you don't want, or you can not use the int0 pin, you can still check the status of the pins every 100ms or so by using the WDT (which is running on the 128kHz oscillator), or use the Simple Sleep library above (I'm sure that is using the WDT as well). As you can see from the datasheet the sleep mode will only use 4μA in sleep mode with the WDT enabled, the 128kHz oscillator about 50μA (at 3V) and the 1MHz around 500μA (but you would only run it at 1MHz for some microseconds when using the sleep mode).

SpenceKonde, 2019-04-12:
Agree @freemovers - though I would use PCINTs - he says "switches" not "switch", so the ability to use PCINTs on more pins would make implementation easier. Could probably just have an empty ISR and have the chip check the switch status when it wakes up...

ideaalab, 2019-04-12:
Thanks for your approach ideas!
This time i think it will be enough simply by reducing oscillator speed. I know i can cut down further if i go to sleep, but will make the code a little bit more complicated, and i want to keep it straightforward for this project, as it will not be mainteined by me, but by others. So better not to trust in their knowledge :)

ideaalab, 2019-04-12:
> you'll have to uninstall the board manager version, and install the github version to use it before the next release

Instead of uninstalling is it possible to just overwrite the boards.txt file with the new one?

SpenceKonde, 2019-04-12:
Oh - and when I talked about a manual install into hardware, I mean the hardware folder in your sketchbook folder (you may need to create it), not the hardware folder under the IDE install location, which should never be modified. See https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

That said, modifying the boards.txt of an installed board manager package would probably work - though I generally don't recommend digging around in installed board manager packages and modifying them. 

The boards.txt for packages installed via board manager is buried somewhere under the Arduino15 folder (I forget the directory structure within Arduino15, but ones you're there it's straightforward to find):
On Mac OS X:
/Users/(username)/Library/Arduino15

On Windows:
C:\Users\(username)\AppData\Local\Arduino15
(note that AppData is a hiden folder)

On Linux:
/home/(username)/.arduino15    (a.k.a. ~/.arduino15)

ideaalab, 2019-05-03:
Hi!
While waiting for next release to include the missing oscillation speeds, ive simply overwritten the boards.txt
Now 128Khz is showing for Attiny24/44/84, but its not showing for Attiny44/84 (optiboot).
I think it must be added.
Additionally, i would change the order of the list. And would place all "internal" options on top. Now its 8Mhz internal at the top of the list, and 128Khz internal at the bottom of the list, with all the external options in between.
Just a suggestion...

SpenceKonde, 2019-05-03:
I dont provide oscillator clock sources slower than 1MHz for optiboot configurations. At 128 kHz, youre looking at 600 baud for the bootloader, which is slow enough that problems

ideaalab, 2019-05-03:
1Mhz is not showing either for **Attiny44/84 (optiboot)**

SpenceKonde, 2019-05-03:
That should be fixed.

SpenceKonde, 2019-05-17:
Also worth noting that the makefile is missing a shitload of entries... I sure hope I have a clean copy of it at home.

commit b471d6a7 (2019-04-12): Add missing watchdog/ULP clock sources - fix #310

As reported in #310, the ATtinyx4 series didn't have 128kHz clock option. On further investigation, it was found that only the x5 series, and the x41 (which has multiple ULP clock settings) had these available. Added this option for all non-optiboot board definitions.
commit 1804597a (2019-05-26): 1MHz bootloaders (fix #310)

Expanded makefile to add 1MHz bootloaders, added binaries
closed 2019-05-26 by SpenceKonde