SpenceKonde/ATTinyCore #311: attiny85 strange I2C problem

nedoskiv, 2019-04-16:
Hello,

First I want to thank you for creating and maintaining this library, It is only one that made my attiny works.

I noticed something strange about I2C communication in slave mode. As my code grow, I2C communication start to fail, also it depend where i gonna place interrupt handlers in code (before main loop, after it ...), noticed that placing it at the beggining make communication works better, but still cause it to fail from time to time.

Here is a topic I posted in arduino forum, which include my source code.
https://forum.arduino.cc/index.php?topic=610182.0

If code goes in lot of else/if statements it take a little bit more time and communication going worse, for example if it goes thro last else if statement:

 //    SSERIAL.println("ext power is OFF");

it is almost failing even in current state of the code.

I'm posting that here, because use similar code on atmega8 where I was filled all flash available  (8k) and there is not such a problem.

SpenceKonde, 2019-04-16:
The fact that the position of the ISR declaration in the .ino file makes a difference is wacky. Are you sure of this? 

Are you able to post code which demonstrates this issue without depending on external hardware so I can try to reproduce it in house? I also need the sketch running on the I2C master that talks to it and reproduces this. 

I would not be surprised if there were issues with I2C slave mode - that functionality has not been well exercised. If I can reproduce your issue, this is a priority for the next release.

nedoskiv, 2019-04-16:
Thank you for quick anwer, code is in arduino forum, but I need to make it on non working state. In the way I posted it it works in most of the cases except in one when go thro longest else if statement in main loop, then it cause some errors.

Master is ESP32 micropython. Hope you have one, or ESP8266 may do the trick too. Let me know which one of them you have.

I will test/prepare working and non woking example for you in 2 days and post links to files here.

nedoskiv, 2019-04-17:
Well I'm glad to inform that problem is not into attinycore library. That what I describe really happen to me, but on a specified device, when I try to reproduce it on breadboard it works fine. So I gonna take a look at power sources etc. Still got no explanation why that happen.

Sorry to bother you for nothing.