SpenceKonde/ATTinyCore #312: EEPROM are not being retained after loading a sketch

jhmluna, 2019-04-22:
Hello.

This is the first issue I open, so please forgive me if I am not following some procedure.
I noticed that although the "Save EEPROM" setting is set to EEPROM retained, every time I load a sketch, the EEPROM is erased.
Is there any other setting that I have to do so the EEPROM is not cleared when programming attiny2313?
I'm using ATTinyCore library version 1.2.3 to program an Attiny2313. And I'm using as programmer an Arduino Nano: Arduino as ISP (ATTinyCore).
Arduino IDE version is 1.8.9.

[image]

liebman, 2019-04-22:
That value is actually a fuse setting.  You need to "burn bootloader" to apply fuse settings for them to be applied.

jhmluna, 2019-04-23:
Thank you very much! It worked.

closed 2019-04-23 by jhmluna