SpenceKonde/ATTinyCore #313: On some versions of Arduino IDE, cached cores exceed MAX_PATH on windows

SpenceKonde, 2019-04-29:
http://forum.arduino.cc/index.php?topic=612754

SpenceKonde, 2019-04-29:
Fixed by ce8cde8d366443a274100abdd3a4a1e741934939 which shortens the names used internally (but not user-facing) for these options.

sleemanj, 2019-04-29:
> http://forum.arduino.cc/index.php?topic=612754

Wrong link?

SpenceKonde, 2019-04-29:
http://forum.arduino.cc/index.php?topic=605303 is the correct link, oops

sleemanj, 2019-04-29:
FWIW

I think this was probably "fixed" (reduced chance of it happening) in 1.8.6 when the shipped arduino-builder was updated to 1.4.0

https://www.arduino.cc/en/Main/ReleaseNotes
https://github.com/arduino/arduino-builder/pull/249
https://github.com/arduino/Arduino/issues/6751 (and issues referencing it)

SpenceKonde, 2019-05-02:
Reverted these changes until I can figure out what to do so that someone syncing from github doesn't get something that won't work with automated builds.

commit 002f02f8 (2019-05-13): Fix #313 - refuse to compile if will hit MAX_PATH

Not an issue on non "modern" cores, because it was the new menu options that pushed it over the line.
closed 2019-05-13 by SpenceKonde