SpenceKonde/ATTinyCore #314: ATTinyCore periodically disappears from Boards Manager

technoblogy, 2019-05-07:
I've added **http://drazzy.com/package_drazzy.com_index.json** to my **Additional Boards Manager URLs** list in **Preferences**, but on a couple of occasions I've gone into **Boards Manager...** from the **Tools** menu and the ATTinyCore entry has disappeared. When I check in **Preferences** the json entry for drazzy.com has also disappeared.

Entering it again and restarting the Arduino IDE then seems to find the ATTinyCore again, without reinstalling it.

I'm using Arduino IDE 1.8.9 on the Mac. Any suggestions?

SpenceKonde, 2019-05-07:
If the json url is no longer being used by the ide, the core will be lost by the ide. If the ide "losing" the url, this is an IDE bug; its out of my control.

As a workaround though, you can do a manual installation of the core, see the installation doc.

closed 2019-05-07 by SpenceKonde

technoblogy, 2019-05-07:
I've got 8 other json URLs in Preferences and it's never happened with one of the other cores, which is strange.

per1234, 2019-05-07:
This could be caused by the server that has http://drazzy.com/package_drazzy.com_index.json being inaccessible when you open Boards Manager. The Arduino IDE downloads fresh copies of all the JSON files from the Boards Manager URLs and doesn't save the existing JSON file, so if the download fails, then you get the behavior you saw. That was happening a lot with the ESP8266 package when they were having server issues.

When it happens, after closing Boards Manager, you should see some information about the download failure in the black console window at the bottom of the Arduino IDE window.

technoblogy, 2019-05-07:
Thanks, David

SpenceKonde, 2019-05-09:
This is mysterious indeed! If you're able to capture the error messages if/when it occurs again, please let me know! 
Nobody else has reported issues, so it seems unlikely that it could be a server issue (and the server has been running continuously without any issues to my knowledge)

technoblogy, 2019-05-09:
I'll try and get an error message next time it happens, but if no one else is having the problem it's probably not worth worrying about. Thanks!