SpenceKonde/ATTinyCore #316: Serial.available() not working on Attiny44

nadarbreicq, 2019-05-13:
Hi !

I'm working on a project with Attiny, and I'm testing the Serial functionnalities. Write functions from Attiny works well, but it seems to have some issues with the Read functions. 

As an example, `Serial.available()` never send anything when something is send to the Serial port.
Is  `Serial.available()` compatible with the X4 series ? And if yes, is there a good way to use it ?

Tanks in advance !

SpenceKonde, 2019-05-13:
This should work.

SpenceKonde, 2019-05-13:
You aren't by any chance sending it multiple characters, and then trying to echo them back as soon as you receive them (ie, while it's still receiving the other characters in the message) are you? The serial on the ATTiny44 is a software implementation - you cannot send and receive at the same time.

nadarbreicq, 2019-05-14:
Ok ! I've got my Serial.available() working now. I was wrong on my connection : I confused the pin AINx with the PCINTx. The wiring of the attinyX4 is a bit confusing for me also because of the "alternative pinout" which I was not aware of. So I've got some kind of differences between my code on the Arduino IDE and on PlatformIO. 

Maybe it should be good to add on the pinout drawing the Serial pins TX and RX that works by default with the library ? Like you did on the pin-mapping on the attinyX41.

Thank you in advance and sorry to bothering you with this "non problem" :)

SpenceKonde, 2019-07-02:
@MCUdude - are you able to add the TX/RX labels to AIN0 and AIN1 pins on 43, x4, x5, x61, and x8 pinout diagrams or should I try to do it myself.
(if there's an obvious way of marking them as being different from normal serial interface - as they're a software implementation with all the issues that entails - that would be good, but I'm not sure what the right way to show that is. Any ideas?

MCUdude, 2019-07-02:
That should be doable! How about a new color and a new item in the list called "SOFTWARE SERIAL"?

SpenceKonde, 2019-07-02:
Maybe SERIAL (SOFTWARE)?

I want to avoid "SOFTWARE SERIAL" because that would lead people to think that those pins were to be used with the SoftwareSerial library.

MCUdude, 2019-07-02:
Is this what you want?

[image]

MCUdude, 2019-07-02:
Sorry, I posted right before I saw your message. I can re-do it

MCUdude, 2019-07-02:
[image]
 Here you go

SpenceKonde, 2019-07-03:
That is *perfect*! Thank you!

When you have a chance, if you could do this for 43, x4, x61, and x8 as well, that would be awesome <3

SpenceKonde, 2019-07-03:
For the x4, can you change "alternative pinout" to "counterclockwise pinout" too?

MCUdude, 2019-07-05:
Here you go!
I exported the whole project, so you'll have to scissor out the ones you need.
[ATTinyCore_pinouts.zip](https://github.com/SpenceKonde/ATTinyCore/files/3364029/ATTinyCore_pinouts.zip)

SpenceKonde, 2019-07-14:
Wonderful, thanks!

When you have a chance can you do this to the x8 series TQFP version too?

Thanks
-Spence

MCUdude, 2019-07-17:
[AttinyX8.zip](https://github.com/SpenceKonde/ATTinyCore/files/3402855/AttinyX8.zip)

Here you go! You might need to crop the picture. All I did was adding the labels and exporting the page without any resizing.

SpenceKonde, 2019-07-17:
Great! Thanks again!

I'm fine with doing the cropping and resizing.

commit 51d65186 (2019-07-14): Pinout diagrams

Fixes #316 except for tiny x8 in TQFP
closed 2019-07-19 by SpenceKonde