SpenceKonde/ATTinyCore #319: Change PWM Frequency on ATTINY85 but keep millis() operational

circuitsirl, 2019-05-22:
I am hoping to get 3 or 4 PWM pins on an ATTINY85 running at 10+ kHz each (20kHz would be ideal).  I would like to use the 8MHz internal clock. This is going to require modifying timer0 which will break millis() in Arduino.  Is there an easy way to go about doing this?  I have poked around a bit in wiring.c but I really do not know what I am doing.  

Thank you for this excellent core and apologies for asking questions in the issues tracker.

sleemanj, 2019-05-22:
In [my fork's distrbution](https://github.com/sleemanj/optiboot/blob/master/dists/README.md), you can choose a No Millis option for the x5 which disables all the `millis()` handling stuff.

circuitsirl, 2019-05-22:
Thanks sleeman but I would really like to keep millis() working.

SpenceKonde, 2019-05-22:
There is no way to do this without great effort - if you need 3 PWM pins (that's all you can get, each timer has 2, but on the '85, two of those are on the same pin, so you get 3) at a custom frequency on the t85, you lose millis() unless you put a great deal of effort into reimplementing millis() to work with the different timer settings; worse still, if you're kicking the frequency up to 20kHz, doing so will result in the system having to handle a lot more interrupts than it otherwise would (80x as many) to update millis. 

I would strongly consider switching to a chip with 3 timers, so you could keep using timer0 for millis; the attiny841 would be an excellent choice, as it's timer1 and timer2 are identical 16-bit timers (the same as timer1 on most chips, not the weird timer1 the '85 has (not that tiny85's timer1 isn't a great peripheral).

closed 2019-05-22 by SpenceKonde

circuitsirl, 2019-05-24:
Thanks Spence, really helpful to know from someone as knowledgeable as yourself on this that it's not worth my time to figure out!  I will look into using the 841.