SpenceKonde/ATTinyCore #320: Attiny 441/841 Pinout Image is Incorrect...

Dreded, 2019-05-23:
Let me start by saying I appreciate the work... 

the IDE defaults to what is called Clockwise Configuration but the pinout on github image is whats called counterlockwise rotation

**Only read below if you feel like reading a rant(you really dont need to**

So after many hours of frustration trying to figure out why my board wouldnt work(that has no LEDs etc) I bought a SOIC to breadboard adapter so I could breadboard my project to find out shortly after putting an LED on every pin that the IDE defaults to what is called Clockwise Configuration but the pinout image is counterclockwise.

srw777, 2019-05-23:
I'll second this.  This has bit me a few times, already, including just yesterday.  I just keep forgetting to file an issue.

SpenceKonde, 2019-05-23:
Shit, the image with both pinouts listed on it isn't up? Oops.

SpenceKonde, 2019-05-26:
Image has been corrected.

closed 2019-05-26 by SpenceKonde

futuranet, 2019-06-19:
Just wanted to report that ATtiny_x41.md has the old counter-clockwise image at the top. Hope this is the right place!

futuranet, 2019-06-19:
also in repo's README.md

SpenceKonde, 2019-06-19:
Ugh. thanks, fixed now.