SpenceKonde/ATTinyCore #321: increase I2C TX/RX buffer size.

nedoskiv, 2019-05-24:
Hello
and sorry for posting my question as a issue. 
I'm using AttinyCore on attiny 84 and it do wonderful job,  but i need to  increase TX/RX buffer for i2c communication (as slave device)

I do not found it in:
C:\Users\ven\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.2.3\libraries\Wire\src
but found something in:
ATTinyCore\hardware\avr\1.2.3\libraries\Wire\src\USI_TWI_Slave\USI_TWI_Slave.h

is there right file to edit in order to change buffer size?

SpenceKonde, 2019-05-24:
That's correct - for I2C on devices with USI hardware (all ATTinyCore supported devices except 88/48,828,841/441,1634), the USI_TWI_Slave and USI_TWI_Master folders contain the settings and options relevant to Wire in slave and master mode. Be sure to use a supported buffer size (see comments in the code), and also be aware that you really don't have very much ram to work with...

SpenceKonde, 2019-05-24:
Closing because this isn't a bug with the core, but I'm not unsubscribing so if you have further questions, feel free to ask them here and I'll see them (just don't want it sitting in the open issue list).

closed 2019-05-24 by SpenceKonde

nedoskiv, 2019-05-25:
Thank you, 
what about list of those devices you mention, where are settings for them?
and few silly questions if you do not mind.
Guess when I increase buffer size it gonna eat from arduinoide dinamyc memory?
and how actually buffer things works, when is it flushed, for example if I split my code like this:
`  
Wire.write("1234560,9,1234569");

Wire.write("1234560,9,1234569");
Wire.write("1234560,9,1234569");`

It still do not flush buffer between wire.write, I assume it write into buffer and it is flushed when device manage to sent it?